dalathegreat/Battery-Emulator #915: TESLA 2024 LFP HVIL Troubleshooting

1ni3shi, 2025-02-24:
Hello,
I'm currently stuck with my installation Tesla Model 2024 3/Y LFP battery with part number 1666969-00-E with Deye inverter SUN29.9K.

I have followed all the documentations I can find here and via BatteryMan's YouTube, however I've exhausted all my knowledge.

1. Contactors closing (I've never had the contactors close before)
Main issue preventing contractors from closing is the HVIL.
I have triple checked all the jumpers and the 60ohms resistor on the X098 pins are seated.

[image]

[image]

[image]

I get two different sets of errors, one with just HVIL (UNKNOWN 10) 
[image]

This was a rare one with HVIL (vehicle open vault) penthouse was closed during these testing
[image]

This one has HVIL (NOT OK) and pyro squib
[image]

2. Deye Inverter cannot see the battery
Followed the Deye inverter instructions both Battery and Inverter are on the same CAN pins with the 120ohms resistor removed
Have tried switching between Lithium mode 00 and 01, however 01 seems to be the correct one as there's no issues in the event

[image]

I've attached some CAN logs if it helps with identifying the issue.
[image]

[image]

I know it's a long one, but thanks in advance!

1ni3shi, 2025-02-27:
Did a few more testing with what I could, 
reseated pyro fuse multiple times, also replaced with a different pyro fuse. results were all the same.

Have a few questions about what I'm seeing, my SOC is at 94.80 for the past 2 months. Is it normal?
I turned on scaled SOC which has a reading of 93.5

Max charge current limited to 14.4A (BMS) both inverter and lilygo are set to 50A.
Is this normal as the
[image]

[image]

Updated the error message from debug.
[image]

If anyone has any experience on this and can share that will be fantastic.

1ni3shi, 2025-02-28:
Another day of further testing,
1. Disconnecting X098 Pin 1 and 3, checked for continuity - passed
Powering up lilygo I will get the Vehicle_Open_Fault
2. Removing the 60ohms resistor Pin 1 and 3
Without any resistor, 60 ohms and 120 ohms resistor, I get the same error message for check high voltage connectors and interlock circuits.

1ni3shi, 2025-03-04:
My conclusion is that the Highland Model 3 battery has a different HVIL circuit setup, with the Drive Unit 60ohms resistor removed from X098 Pins 1 and 3. 
Have tried modifying the codes to bypass HVIL to see if contactors can close, but with my limited ability I was unable to resolve the issue.

The Battery Emulator codes might need to be updated/modified to enable further testing, as to external connections the Drive Unit has been moved into parts of thee Right Body Control according to the latest Tesla Schematics.

hotmail6, 2025-03-07:
Hey buddy, 
I'm at the same point at the moment (HVIL fault Unknown 10). Battery from a 2024 model 3.
Have you added a capacitor (it may not be tied to the HVinterlock.
I have replaced the blown pyro fuse. (I checked the resistance of the squib before installing and it was 2.2 Ohms which should have been good but getting a pyro squib not connected error now too(I had prev. just a twisted piece of copper wire and a 2 Ohm resistor while I was waiting for a new fuse))

hotmail6, 2025-03-07:
I've got a couple more days before I have to go back to work for a couple of weeks. I paid a guy to reset my inverter tech login yesterday as they lock them out when they are installed so that the parameters can't be changed (like adding a battery). I'll carry on checking a few things today I must have the highlander style battery too. I'll stretch a couple of wires to the motor to make sure it's not something to do with that. They may have gotten a little tricky with this model.
Worst case I imagine the contactors just require 16V to close. But I really don't want to have to do that.
I am an electrician but with the lack of wiring diagrams etc. I'm feel a bit unsafe working under the lid. I notice when replacing the pyro fuse the is a little spark when bolting it down.
I didn't see a penthouse open fault in your code so I assume you have had the lid bolted down each time. The penthouse lid switch runs between the bolts either side of the coms and 16v cable plug.

1ni3shi, 2025-03-07:
I have the original 16V lithium battery for the highland, its currently sitting around 14.5V which is in line with the specs of 85%.
So I do not believe you need 16V to close the contactors.
I've gone through many schematics comparing SOP8 and SOP7, you can check it out here
https://service.tesla.com/docs/Model3/ElectricalReference/

Yes, there's always sparks when disconnecting and connecting the pyro fuse, just make sure you follow the safety standards.
I've tried with penthouse lid on and off (grounding the 2 screws next to the X098, all the same errors.

CAN signal is the only thing I'm unable to troubleshoot.
The highland BMS could be expecting a signal from the drive unit or another HVIL to confirm HVIL OKAY.

hotmail6, 2025-03-08:
[image]

I didn't have the squib fault when I had a 2 Ohm resistor across the squib connector.
Ah yeah those linked diagrams. Had to VPN to check those initially. Unavailable in Australia. I swear they have made them difficult to sift through on purpose.

hotmail6, 2025-03-08:
[image]

This is what I had prior to replacing the pyro.

1ni3shi, 2025-03-08:
Yeah, so I’ve noticed the pyro squib error is there more often than not. It’s probably not random but I’ve had occasions where it wasn’t there without changing any hardware. 
The schematic are accessible in Australia (im also in Australia), I’m just glad to know I’m not the only person working on a highland battery pack.

hotmail6, 2025-03-08:
Rgr, between the two of us we should be able to get them going.

1ni3shi, 2025-03-08:
Well, I’m at the point where I’m not sure what else can be done. I’ve been stuck chasing around the software side of things, trying to see if I can manipulate CAN signals but it’s a little out of my league. I’ve enlisted the help of DeepSeek to trouble shoot the CAN logs and error messages.

hotmail6, 2025-03-08:
https://www.linkedin.com/pulse/model-3-bms-main-control-board-disassembled-keven-chen/
Have a look at this one. Mostly just pictures and plugs. I'm trying to find what they are calling the P2 plug in the tesla docs for a pinout.
I pulled my board out to see if they had added any phisical interlocks but it largely looks the same.

hotmail6, 2025-03-08:
https://service.tesla.com/docs/Model3/ElectricalReference/prog-233/interactive/html/index.html?page=29&
This too for IL wiring

1ni3shi, 2025-03-08:
That’s inside the penthouse, I’ve noticed touched anything inside the penthouse. The only HV connectors I have access to is the RWD (using original cable), AC (jumper) and the 2 charger sockets (jumper). 
Supposedly the HVIL out is 2.4V and HVIL in is 1.2V, but I have no way of measuring the HVIL circuit. 

When I measured the HVIL circuit opened, it was 4.8v at both the charging ports and 8.6v and the AC connector.

hotmail6, 2025-03-08:
The interlock circuit he has labelled on the side has 5 resistors? there should be 5 phisical interlock circuits. But I recon your on the right track with the software side. Where is the actual programming code in the Arduino all I can see is what looks more like parameters.

1ni3shi, 2025-03-08:
Where are you seeing the 5 resistors and 5 physical interlocks? From the Tesla schematics?

Code is here
https://github.com/dalathegreat/Battery-Emulator/blob/main/Software/src/battery/TESLA-BATTERY.cpp

hotmail6, 2025-03-09:
This one from the linkedin link.

[image]

1ni3shi, 2025-03-09:
I missed that link earlier, if you find anything useful please enlightenment.

hotmail6, 2025-03-09:
https://tesware.net/alerts/model3/BMS_a071_SW_SM_TransCon_Not_Met
This one to enter your ERRORs into. Gives a slightly better but not all that helpful explanation.
I am feeding the system with 13V from a bench lab at the moment I'm back down to...
      26.074 Event: CAN message failed to send within defined time. Contact developers, CPU load might be too high.
      27.001 ERROR: Check high voltage connectors and interlock circuit! Closing contactor not allowed! Values: 
      27.001 ERROR: BMS_a052_SW_PCS_MIA
It's interesting that you aren't getting this fault care to share some pics of your setup. I still have the 60 Ohm resistor between on pins 1 and 3, you said you had moved yours?

hotmail6, 2025-03-09:
I have the capacitor across the ac feed. The voltage is right the the
capacitance is in Mega Farads, shouldn't matter but I have the correct one
on order. I found the if I try to apply the 16v to the dc dc out put I get
squib fails and dclink overvoltage etc. Got to head out for lunch I'll
chase up the resistor wiring when I get back.

1ni3shi, 2025-03-09:
I have put back the 60 ohms resistor back between pins 1 and 3. But from testing there’s no difference in terms of error message. No resistor, 60 and 120 ohms.

hotmail6, 2025-03-09:
Try a 120 Ohm resistor as a terminating resistor between the green and black twisted pair. Your errors are a little different to mine.
The twisted pair is for can bus and having an open loop without the terminating resistor will mean nothing else will be read on that part of the network.
It hasn't fixed my problem but has made mine start looking for the wifi now for some reason.
And you have your code set to look for the PYLON_LV_CAN line 55 of user_settings.h as battery man did until it is hooked up to the inverter?

hotmail6, 2025-03-09:
F that can leg goes to the charging port ecu... i threw mine out.

1ni3shi, 2025-03-09:
I have mine setup to the inverter already, my inverter runs on BYD_CAN and I have 120ohms resistor on the battery CAN lines.
The main error we have are the same, HVIL, pyro squib, arc, HW contactor and DC Link over volt.
The most important one is still to get HVIL OKAY so we can test the others.

hotmail6, 2025-03-09:
If you tell it that you are PYLON it doesn't require can of modbus back from the inverter coms. It's just something for fault finding to rule out.
I've got the gen24 10kw I think it's supposed to be BYD modbus for me I suppose I could try hooking it into the inverter.

Agreed' the HVIL killin us.

Did you manage to reset the 16V li-ion aux battery or are you bypassing it's BMS like me.  (car was already in pieces before I found how to do the force close mosfet from service window)

1ni3shi, 2025-03-09:
I’ve tried using the fake test battery, it’ll just receive CAN from the BMS. But it’s not very useful as we need to send CAN instructions for feedback. 

What’s the purpose of resetting the 16V lithium battery?
Mine was disconnected from the battery when I initially got it and it’s usually disconnected after I’m done testing.

hotmail6, 2025-03-09:
I can't use the batterys built in BMS to charge or discharge it's locked out.

[image]

1ni3shi, 2025-03-09:
I use the 16v lithium to power my Lilygo, so I know it works. I’ve never charged it as the contactors needs to be closed in order for the PCS to charge the 16v lithium via the 12V

1ni3shi, 2025-03-10:
are you able to share with me your CAN logs for the following IDs

RX0 20A [6] C9 11 09 92 10 00
RX0 3AA [8] 01 00 00 00 00 00 00 00

hotmail6, 2025-03-10:
Nope you are going to have to do better than that.

1ni3shi, 2025-03-10:
So login to the web server the default ip is 192.168.10.150 unless you’ve changed it. 
Once there just scroll down to the bottom and there will be a few different buttons, one of them is can log. 
Press on that and at the top you will need to start logging and refresh the page

hotmail6, 2025-03-10:
You look to be running yours off of a phone or tablet. I'm using arduino on windows and going into a browser to type 192.168.10.150 goes nowhere.

1ni3shi, 2025-03-10:
In User_Settings.cpp goto
IPAddress local_IP(192, 168, 10, 150);
IPAddress gateway(192, 168, 10, 1);
IPAddress subnet(255, 255, 255, 0);

Change it to match your home network.

In User_settings.h goto /* Connectivity options */
and enable #define WIFICONFIG

hotmail6, 2025-03-10:
Yeah no dice.
I've match to my gateway and subnet by checking ipconfig locally, enabled wificonfig, connected to the battery emulator network and entered the 192.168.10.150.
I am out of time for now. I'm out for a couple of weeks I'll try again when I'm back.

closed 2025-03-13 by 1ni3shi