dalathegreat/Battery-Emulator #919: [Feature request] using gely geometric c 2023 battery to solar system storage

Shalabnyar, 2025-02-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I need to use HV battery from ev car gely geometric c 2022, it's a 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
370v
17 string
Li io
3.6v block  battery each 102s
I want to connect it to my inverter using its own bms

If someone can help?
[image]
[image]
[image]
[image]
[image]

**Additional context**
Add any other context or screenshots about the feature request here.

alaatamim1996, 2025-03-07:
I hope you do this 🥹

alaatamim1996, 2025-03-09:
I also did that and set the maximum charge to 420 volts and the minimum discharge to 300 volts. Is this okay?

Shalabnyar, 2025-03-09:
No
440 max
360min

alaatamim1996, 2025-03-09:
As far as I know, each cell can be charged to 4.2 volts, which is equal to 4.2 x 102 = 428. Each cell can be discharged to 2.8 volts, which is equal to 102 x 2.8 = 285 volts, taking into account safety standards so that it is not fully charged and fully discharged. Charging is done to 420 volts and discharging to 300 volts.

Shalabnyar, 2025-03-10:
شو رايك تبعثلي واتس
0502963367