dalathegreat/Battery-Emulator #922: Tesla Model 3 battery not charging

Jaywatts, 2025-02-26:
I've hooked my model 3 battery up to my Fronius Gen24 inverter and coms seen ok.
The issue I have is the contractors close and then open again after about 12 seconds. I believe it's because the battery power is low and deviation between cells is high.
I need the contractors to stay closed so that I can charge the battery and balance the cells. Would anyone have a work around to allow this because the more I muck around with it the lower the voltage and the higher the deviation gets?

[image]
[image]
Thanks heaps.

Jaywatts, 2025-02-26:
Would I be able to connect the high voltage inverter leads to the battery side of the contactors until the voltage in the battery comes up?

dalathegreat, 2025-02-26:
Hi @Jaywatts 
Wow that battery is LOW. You need to charge it up right away to prevent damage to the cells.
Start slow, reduce power.

To be able to get the battery to keep contactors engaged, you will need to modify the Battery-Emulator/Software/src/battery/TESLA-BATTERY.h file

Modify the cellvoltage min limits, so they wont interfere with the charging
MIN_CELL_VOLTAGE... 

[image]

Good luck!

Jaywatts, 2025-02-26:
Great, thank you very much. I'll give it a go.

Jaywatts, 2025-03-04:
Good news, I was able to keep the contractors closed and bring battery voltage up. The cells have also equalised nicely. Thanks Dala! 😊

[image]

[image]