dalathegreat/Battery-Emulator #931: Incorrect SOC Tesla 55 kWh (LFP)

Dima200-RB, 2025-03-03:
First of all, I would like to express my heartfelt gratitude to the founder of this project, as well as all its participants! The system is working excellently. I launched it in September 2024.
However, after the winter period, I have a suspicion that the battery has started to inaccurately determine the State of Charge (SOC). This might be related to the fact that, out of ignorance, I did not charge it to 100%, but instead maintained the charge at around 50%.
Now, when the charge exceeds 71–72%, a cell imbalance begins (this did not happen before). When discharging to 0%, the voltage on the cells remains around 3.24–3.25 V (although it should be around 3 V). This makes me think that the battery's BMS is incorrectly determining the SOC.
I tried discharging the battery below 0% once: I reduced the charge level to 7%, enabled the "Manual LFP balancing" function (thereby slightly "tricking" the inverter), and monitored the cell voltage during discharge. I discharged to approximately 3.2 V per cell, which consumed about 12 kWh. After that, I started charging.
The SOC showed 1% after about 7–8 kWh had been fed into the battery. And after 13 kWh had been fed in, the SOC only rose to 8%, which seems strange to me.
Has anyone encountered a similar problem? Could you please advise me on where to look for a solution?

1ni3shi, 2025-03-10:
Can you please share what your CAN ID 20A and 3AA are in the logs when your contactors are closed?
RX0 20A [6] C9 11 09 92 10 00
RX0 3AA [8] 01 00 00 00 00 00 00 00

Thanks

Dima200-RB, 2025-03-12:
> Можете ли вы поделиться своими CAN ID 20A и 3AA в журналах, когда ваши контакторы замкнуты? RX0 20A [6] C9 11 09 92 10 00 RX0 3AA [8] 01 00 00 00 00 00 00 00
> 
> Спасибо

RX0 20A [6] F6 15 09 92 18 01
RX0 3AA [8] 00 00 14 00 01 02 00 00

1ni3shi, 2025-03-13:
Thank you so much. With that I was able to get my 2024 Highland battery contactors to close.

closed 2025-11-23 by dalathegreat