dalathegreat/Battery-Emulator #944: Sofar HYD5-20KTL 3PH + Tesla model 3/Y battery CAN comunication issue

pyeco64, 2025-03-06:
Hello,

First of all, a big thank you for this amazing project enabling the second-life use of automotive batteries!

I am currently working on a home solar power project using a Sofar HYD20KTL 3PH inverter and a Tesla Model 3/Y battery. I have successfully connected the battery to the LilyGO and can read battery data. However, I am experiencing an issue with communication between the LilyGO and the inverter.

I am getting the following error:
"can_native failed to transmit, or no one on the BUS to ACK the message."

I would appreciate any advice on what might be causing this issue. I have checked the connections and configuration, but I haven’t been able to resolve it yet. Any help or suggestions would be greatly appreciated!

Once again, thank you for all your hard work on this project!

Best regards,
Peter

1ni3shi, 2025-03-06:
Without a bit more information on your wiring and setup,
1. Lilygo is set to PYLON_CAN
2. Both the Battery and Inverter CAN-L and CAN-H are together on the Lilygo
3. Have you removed the 120Ohms resistor on the Lilygo
4. Have 120ohms resistor over the battery CAN-L and CAN-H wires

pyeco64, 2025-03-06:
> Without a bit more information on your wiring and setup,
> 
> 1. Lilygo is set to PYLON_CAN
> 2. Both the Battery and Inverter CAN-L and CAN-H are together on the Lilygo
> 3. Have you removed the 120Ohms resistor on the Lilygo
> 4. Have 120ohms resistor over the battery CAN-L and CAN-H wires

1. yes
2. yes
3. yes
4. yes i have 61omh resistance when everything off  on can-l and can-H

[image]

dalathegreat, 2025-03-06:
Pylon CAN uses extended CAN IDs, and Tesla uses normal CAN. Could be that this crashes the bus in a strange way.

I suggest adding another CAN channel, dedicated only for the inverter.

- You can [add an isolated MCP2515 CAN channel](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-add%E2%80%90on-(MCP2515))
- You can [add an isolated MCP2518 CANFD channel, and run it in classic CAN mode](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN%E2%80%90FD-add%E2%80%90on-(MCP2518FD))
- You can use the [Stark CMR](https://github.com/dalathegreat/Battery-Emulator/wiki/Hardware:-Stark-CMR) hardware
- You can use a [CAN filter](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-filter-hardware) between inverter and the rest of the system

pyeco64, 2025-03-06:
Hello,

Used MCP2518FD Pro, getting this mesage on events.

[image]