dalathegreat/Battery-Emulator #951: Tesla Model 3 Battery Manual balancing

Jaywatts, 2025-03-07:
In Manual balancing mode my charging stops @ 100 SOC and Emulator says 
"Battery discharging! (Inverter limiting)" with battery voltage at 371v.
Do any other settings require adjustment?
- Tesla Model 3 LFP
- Lilygo
- Fronius Gen 24 Plus

1ni3shi, 2025-03-10:
Can you please share what your CAN ID 20A and 3AA are in the logs when your contactors are closed?
RX0 20A [6] C9 11 09 92 10 00
RX0 3AA [8] 01 00 00 00 00 00 00 00

Thanks

paulcristianmarcus, 2025-03-19:
Hi! I have the same problem but at 99.7 SOC
Can you share your SOH. I have the same setup

1ni3shi, 2025-03-19:
I think that’s normal, it’s working by design. 

[image]