dalathegreat/Battery-Emulator #960: [BUG]   Can buffer full event with mcp2515 addon board

ErikssonPer, 2025-03-10:
Fresh download of version 8.8.0
Lilygo with can addon mcp2515 board
Compiled for Growatt hv inverter and pylon battery

[image]

Have CAN buffer full event but the system is working  like normal when I check can traffic, it both sends and recieves data from inverter and bms and it's currently charging.

This the first time to use addon board for me, had 2x lilygo's before, which worked without events.

dalathegreat, 2025-03-16:
Hi @ErikssonPer , do you have the Growatt HV inverter, or the Pylon Battery attached to the MCP2515 board?

ErikssonPer, 2025-03-16:
> Hi [@ErikssonPer](https://github.com/ErikssonPer) , do you have the Growatt HV inverter, or the Pylon Battery attached to the MCP2515 board?

Inverter on MCP2515

dalathegreat, 2025-03-16:
Good catch, I have submitted a pull request to fix this. We are sending too many CAN messages at once which makes the buffer overflow.

You can test this PR: https://github.com/dalathegreat/Battery-Emulator/pull/980

Direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/growatt-CAN-load.zip

Let me know if it works better! :)

ErikssonPer, 2025-03-17:
> Let me know if it works better! :)

Tested and can confirm it is fixed.

Outstanding piece of software!

dalathegreat, 2025-03-17:
Thanks for testing, the PR is now merged to main!

closed 2025-03-17 by dalathegreat