dalathegreat/Battery-Emulator #963: [BUG] LilyGo resets on whatever command is sent via MQTT

ruffieuxh, 2025-03-11:
**Describe the bug**
I successfully publish BE/info, events and status topics to my MQTT broker and use them in my node-red installation.
However, when I publish any commands (tried PAUSE, RESUME and BMSRESET), **using QOS 0**, the battery emulator always **performs a reset** as per attached screenshot.

The same problem I already had with 8.6.0

**To Reproduce**
Publish commands  PAUSE, RESUME or  BMSRESET to BE/command/

**Expected behavior**
Published command executed as per command published

**Screenshots**

[image]

[image]

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files 
  [USER_SETTINGS.zip](https://github.com/user-attachments/files/19180168/USER_SETTINGS.zip)

- Software version: 8.8.0

- Battery used: NISSAN_LEAF

- Inverter communication protocol: BYD_MODBUS

- Hardware used for Battery-Emulator: HW_LILYGO

- CONTACTOR_CONTROL: yes

- MQTT: yes

dalathegreat, 2025-03-16:
Hi @ruffieuxh , can you try the newly released 8.8.1 version? It fixes a MQTT watchdog reset

https://github.com/dalathegreat/Battery-Emulator/releases/tag/v8.8.1

Let us know if the issue still happens with 8.8.1!

ruffieuxh, 2025-03-16:
Hi @dalathegreat ,

Yes it is fixed! When sending any payload (i.e. a node-red timestamp) to the topic BE/command/keyword (i.e. "PAUSE") the reaction is as expected.

**However,** if you - by mistake or misunderstanding - publish the keyword (i.e. "PAUSE") **as a payload** to the topic **BE/command**, it still crashes. Maybe you want to add a check there. Alternatively you may want to clarify this in the wiki.

Many thanks for the update!

Heinz

dalathegreat, 2025-03-16:
@No-Signal , thoughts on the payload publishing?

StarkJohan, 2025-03-21:
The mqtt_message_received function definitely should get some filtering added at some point.

StarkJohan, 2025-03-21:
> PR for this bug is ready for review
> 
> [@StarkJohan](https://github.com/StarkJohan) could you expand what you are thinking to make sure I understand. Currently the mqtt_message_received function checks that the incoming topic is a match before any action is taken. This check was broken and so shorter than expected topics could cause unexpected behaviours but the above PR resolves that.

My bad, I read the code quick and wrong on my phone earlier. I see now that there's only action on the exact match on the string.

PR #989 (2025-03-24): MQTT Crash bugfix
Updating mqtt topic processing to incorrect topic matching when a shorter topic than expected is used which in some situations could cause a crash.

Fixes #963