dalathegreat/Battery-Emulator #965: [Feature request] GoodWe ET 30KW (29.9Kw) Dual Battery Ports

craigcurtin-dev, 2025-03-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

The GoodWe ET 30Kw Hybrid Inverter has dual Battery ports - both of which are HV and are limited to 50amps throughput

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A method to Daisy Chain (or otherwise) the Double Battery solution such that they are reporting to the GoodWe as being on port 1 or Port 2

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Presumably having watched some of Dalas other videos with Dual Nissan Leaf batteries they are being connected in Parallel over a shared bus and into a single port. The GoodWe removes this issue (and indeed the inherent inrush current issues with voltage differential in two battery packs, however there will be a need to daisy chain the CAN bus  with different IDs so the inverter can differentiate each one.

**Additional context**
Add any other context or screenshots about the feature request here.

I am currently waiting on stock of the inverter in my country and intend to hook 2 x Tesla M3 batteries to it

Craig

StarkJohan, 2025-03-12:
The obvious presumptive solution is to use the BE as is with two completely separate systems. I.e. one BE and battery for each battery port.

craigcurtin-dev, 2025-03-12:
Thanks for the quick response - the only issue is that the inverter only has a single CAN CAN port for battery communication - can the emulator be set to different addresses if i had more than one and dasiy chained them to the same bus ?

StarkJohan, 2025-03-12:
As two separate systems is an option on the inverter there must be a solution in place. The best way to get this properly setup is to look at the can bus on an operating system to figure out how the can addressing is done. 
We could try setting up a dual BMS system in the inverter app to see if this alters the messages sent from the inverter.

StarkJohan, 2025-03-12:
I have one of these systems myself (25kW-ET) so I would like to see this working in the future. I've been thinking about testing this anyway so I'll see if I get some time to have a look at the CAN bus if you don't beat me to it.
Original two BMS system logs would be optimal.

craigcurtin-dev, 2025-03-12:
Yes reviewing the documentation last night and extrapolating from their examples of using multiple Lynx batteries - they have a single CAN port - and each battery stack has a CAN in and CAN out (and what appears to be a CAN Bus terminator as well) - refer to here around page 60 https://en.goodwe.com/Skippower/downloadFileF?id=2486&mid=60. So presumably the Lynx units have a way of either setting a can bus address per unit - or they have a seperate canbus on the in and out ports (which would seem strange) 

Interestingly in the packing list for the 25K and above they note that they include two CAN comms cables (but it definitely appears that there is only a single CAN port on the inverter) I have not had time to dig into the BYD HVM/HVS literature to see how they connect multiple packs together for communications 

I will not be moving into our new house until August and will be getting the inverter then to make this all happen

Craig

StarkJohan, 2025-03-13:
@craigcurtin-dev You got me confused there for a while as you seemed to be so certain about the number of BMS connections :-) I had to double check and I can confirm that the ET 15-30kW units has two separate BMS connectors marked BMS1 and BMS2. I'm currently using BMS1 connected to the emulator and both battery connector pairs hooked up in parallell to the battery for maximum power utilization.

From the instruction manual: 

[image]

craigcurtin-dev, 2025-03-14:
Aah - i have not seen that in all the manuals i have read so looks like this is a non issue then - sorry for the waste of time

Craig

StarkJohan, 2025-03-14:
I think the request may as well remain open until we know that it actually works, so no worries. You'll probably be the first to try the setup.

dalathegreat, 2025-05-05:
Closing this, added note in Wiki https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-GoodWe#note-on-double-battery-ports

closed 2025-05-05 by dalathegreat