dalathegreat/Battery-Emulator #966: [BUG]  Double Pylon Battery - Compilation Error

darek38wien, 2025-03-12:
I have a problem compiling your program. I would like to connect two identical Pylon HV CAN batteries to a Sofar CAN inverter. The batteries are new, but unfortunately, they do not have an output for parallel operation. Unfortunately, there are errors in the software during compilation. Could you fix it? I really care about this.

C:/Users/darek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\darek\AppData\Local\arduino\sketches\EA38082371C0F0B81297CAA9DDD7C62B\sketch\Software.ino.cpp.o:(.literal._Z9core_loopPv+0x18): undefined reference to `update_values_battery2()'
C:/Users/darek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\darek\AppData\Local\arduino\sketches\EA38082371C0F0B81297CAA9DDD7C62B\sketch\Software.ino.cpp.o: in function `core_loop(void*)':
C:\Users\darek\Dropbox\Komputer\Desktop\Battery-Emulator-8.8.0\Software/Software.ino:250:(.text._Z9core_loopPv+0x54): undefined reference to `update_values_battery2()'
C:/Users/darek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\darek\AppData\Local\arduino\sketches\EA38082371C0F0B81297CAA9DDD7C62B\sketch\src\communication\can\comm_can.cpp.o:(.literal._Z25map_can_frame_to_variableP9CAN_framei+0x0): undefined reference to `handle_incoming_can_frame_battery2(CAN_frame)'
C:/Users/darek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\darek\AppData\Local\arduino\sketches\EA38082371C0F0B81297CAA9DDD7C62B\sketch\src\communication\can\comm_can.cpp.o: in function `map_can_frame_to_variable(CAN_frame*, int)':
C:\Users\darek\Dropbox\Komputer\Desktop\Battery-Emulator-8.8.0\Software\src\communication\can/comm_can.cpp:324:(.text._Z25map_can_frame_to_variableP9CAN_framei+0x6b): undefined reference to `handle_incoming_can_frame_battery2(CAN_frame)'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

[image]

dalathegreat, 2025-03-12:
Hi @darek38wien ,
Good idea, I have implemented this in pull request https://github.com/dalathegreat/Battery-Emulator/pull/967

You can download it directly and test with this direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/double-pylon.zip

Let me know how it goes!

darek38wien, 2025-03-12:
Thanks a lot for the quick response. For now, I have successfully compiled and uploaded the software to the Lilygo board. Once I’m ready with the battery and inverter installation, I will test everything and let you know how it went.

Greetings from Vienna!

closed 2025-03-27 by dalathegreat