dalathegreat/Battery-Emulator #968: [Feature request] Double Battery support for RJXZS_BMS

papa-alo, 2025-03-13:
[image]

Trying to compile double battery with RJXZS_BMS, and getting Compilation error: exit status 1

Support for RJXZS double battery would be a great thing to have.

PresidentOfLatvia, 2025-03-13:
I added the double battery code to the RJXZS.cpp using [#966 ](https://github.com/dalathegreat/Battery-Emulator/issues/966) as template.

closed 2025-03-13 by PresidentOfLatvia

dalathegreat, 2025-03-13:
Double RJXZS is not supported, since only the Native CAN port can go down to 250kbps. We do not have support in the ACAN / FD library to switch down the baud rate

To get support for this, refactoring of the ACAN2515 library is needed, with an added configuration option