dalathegreat/Battery-Emulator #970: Possible Hardware: ESP32-CAN-X2

yngndrw, 2025-03-14:
I was just wondering if anybody has looked into or used the [ESP32-CAN-X2 board from Autosport Labs](https://www.autosportlabs.com/product/esp32-can-x2-dual-can-bus-automotive-grade-development-board/) as it has dual CAN and looks like it might be ideally suited to the project?

I can try it myself when I start on my own setup, but there's no point in me repeating effort if somebody else has already looked into it so I thought I'd ask first!

dalathegreat, 2025-03-14:
Hi @yngndrw , that board uses the ESP32-S3 CPU, which is NOT compatible with this project. It has to be a standard ESP32. Same goes for the RP2040, massive amount of effort to get it ported. But this is open source, and anyone is free to have a go at it :)

yngndrw, 2025-03-14:
From memory, I don't believe that much changed on the newer ESP32-S3 from a code point of view so I don't think it would be too bad to port over. The hardest part will be structuring it so that it can compile for either.

The RP2040 is more difficult, the largest thing is of course the wifi but I don't think the porting would be that bad - Again the infrastructure is likely the harder part.

I'll probably give the ESP32-CAN-X2 a go when I do my setup and see where I get to!

closed 2025-06-17 by dalathegreat

Luca-Timo, 2025-10-21:
Hi @yngndrw,
for my own use I (and Claude for that matter) tried to port Battery-Emulator to ESP32-S3 last weekend. I can not guarantee for anything regarding function and or safety.

The Fork currently compiles, uploads to our custom esp32s3 hardware and the web-page config seems to be working. I had no time for testing until now.

link to my repo: https://github.com/Luca-Timo/Battery-Emulator

I hope this might help you :)