dalathegreat/Battery-Emulator #972: [BUG] "The board was reset due to an exception or panic. Inform developers!" upon startup

uitgeslapen, 2025-03-14:
**Describe the bug**
Was running 8.5.0, upgraded to 8.8.0 via OTA. Noticed the reset after upload took longer as expected, the Lilygo led was yellow. After that MEB battery didn't close contacters. And event log reported "The board was reset due to an exception or panic. Inform developers!"

**To Reproduce**
Upgrade to 8.8.0 using my setup file and hopefully it reproduces...

**Expected behavior**
No "The board was reset due to an exception or panic. Inform developers!" error

**Screenshots**

**Version and settings (please complete the following information):**

[USER_SETTINGS .h.txt](https://github.com/user-attachments/files/19254825/USER_SETTINGS.h.txt)

- Software version: 8.8.0
- Battery used: MEB
- Inverter communication protocol: BYD_CAN
- Hardware used for Battery-Emulator: LILYGO [HW_LILYGO/HW_STARK/HW_DEVKIT]
- CONTACTOR_CONTROL: NO, but am using precharge
- MQTT: No

dalathegreat, 2025-03-18:
Got this info on Discord, but I have not been able to reproduce the crash yet
```

       0.063 Battery emulator 8.9.dev build Mar 17 2025 19:27:35
       0.067 Event: The board was reset from a power-on event. Normal operation
E (356) task_wdt: esp_task_wdt_init(515): TWDT already initialized
       0.076 Battery BMS state changed to: DISCONNECTED
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x0000001d  PS      : 0x00060130  A0      : 0x800eaf30  A1      : 0x3ffbd270  
A2      : 0x3ffcadb8  A3      : 0x3ffdb450  A4      : 0x3ffbd278  A5      : 0x00000001  
A6      : 0x3ffcb4a0  A7      : 0x3ffdb430  A8      : 0x800eaeca  A9      : 0x00000001  
A10     : 0x0000006e  A11     : 0x3ffdbda0  A12     : 0x0000001d  A13     : 0x00000000  
A14     : 0x3ffb5a7c  A15     : 0x3ffcad1c  SAR     : 0x00000017  EXCCAUSE: 0x00000014  
EXCVADDR: 0x0000001c  LBEG    : 0x4008a67c  LEND    : 0x4008a692  LCOUNT  : 0xffffffff  

Backtrace: 0x0000001a:0x3ffbd270 0x400eaf2d:0x3ffbd350 0x4008f50e:0x3ffbd370

ELF file SHA256: b50b1e193

Rebooting...
```

dalathegreat, 2025-03-18:
Crash seems to be related to Wifi handling, since it is on Core1 (Connectivity Core)

uitgeslapen, 2025-03-19:
Oh,...I did have to upgrade/reinstall the wifi library. It was complaining during compilation it was missing the wifi.h file after which I de- an reinstalled it. The error during compilation went away. I don't know if that is the moment the crashes started.

uitgeslapen, 2025-03-20:
I have removed alle files and have re-installed everything starting with Arduino IDE. The crash reproduces.

`E (450) task_wdt: esp_task_wdt_init(515): TWDT already initialized
       0.112 Battery BMS state changed to: DISCONNECTED
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x0000001d  PS      : 0x00060d30  A0      : 0x800ed470  A1      : 0x3ffaf710  
A2      : 0x3ffca498  A3      : 0x3ffd73c0  A4      : 0x3ffaf718  A5      : 0x00000001  
A6      : 0x3ffcad70  A7      : 0x3ffd73a0  A8      : 0x800ed40a  A9      : 0x00000001  
A10     : 0x0000006e  A11     : 0x3ffd9620  A12     : 0x0000001d  A13     : 0x00000000  
A14     : 0x3ffb95f0  A15     : 0x3ffca3fc  SAR     : 0x00000017  EXCCAUSE: 0x00000014  
EXCVADDR: 0x0000001c  LBEG    : 0x4008a848  LEND    : 0x4008a85e  LCOUNT  : 0xffffffff  

Backtrace: 0x0000001a:0x3ffaf710 0x400ed46d:0x3ffaf7f0 0x4008f682:0x3ffaf810

ELF file SHA256: 0d638200d

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16492
load:0x40080400,len:4
load:0x40080404,len:3524
entry 0x400805b8
__ OK __`

uitgeslapen, 2025-03-20:
Re-installed everything again, deleted all files and didn't autoinstall the boad. Seems to work now without error.  Closing the issue and will

mvgalen, 2025-03-23:
Some observations. It occurs at random. Not every boot. 
What I notice is that there are no "Connecting to Wi-Fi..." / "Wi-Fi connected. RSSI: " lines in the log above.
So, whatever happens is happening before wifi is connected.

uitgeslapen, 2025-03-29:
Upgrade to 8.9 today on Stark CMR.
After OTA upgrade, the webpage rebooted and was back up with normal delay. My battery is powered off during OTA to prevent BMS errors.
Removed 12V from Stark and re-engaged emergancy off for battery powering 12V again.

After this the Stark started rebooting in the same loop again, left it for 5 minutes (to get the HW to move it back to 8.5.0 via USB) and I succesfully booted. Leaving it at 8.9 for now. It seems that after reboot the recovery from this bootloop is faster seconds iso minutes. Will experiment with Stark precharge this weekend hence will do a lot of restarts, will leave the latest release on to see how it behaves.

dalathegreat, 2025-03-31:
- If I enable DEBUG_VIA_USB, with ESP32 3.2.0, the board reboots constantly with CPU_RESET ❌
- If I enable DEBUG_VIA_USB, with ESP32 3.1.3, the board core panics with Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. ❌
- If I run the program without DEBUG_VIA_USB, all is fine ✅

uitgeslapen, 2025-04-01:
I do not have DEBUG_VIA_USB enabled. I did have DEBUG_VIA_WEB, but even when disabling I constantly get CPU_RESETs.
If you want I can do some troubleshooting, battery is not working anyway :-(

dalathegreat, 2025-04-06:
@uitgeslapen , Think I found the bug, fix is available in PR 1037, direct download link for testing: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/crash-core1.zip

closed 2025-04-06 by dalathegreat