dalathegreat/Battery-Emulator #974: StarkHW GPIO25 which has a relay and not a FET is used to control the precharge which will not work

uitgeslapen, 2025-03-15:
https://github.com/dalathegreat/Battery-Emulator/blame/ebf8aa454825246fedeb1ea74c27f3debca30bbc/Software/src/devboard/hal/hw_stark.h#L51-L51C25

StarkHW used GPIO25 for precharge control, which is a PWM signal. However the HW has a relay on this output and not a FET like other outputs.
Proposal is to move for stark Precharge pin to either GPIO33 or GPIO23 and prevent them to be used together with contactor control or BMS switching, which is not needed when using precharge controle

StarkJohan, 2025-03-15:
GPIO25 works just fine for precharge which in most cases is not a PWM signal. 
However, in case of using PWM and precharge combined, you are correct that it's not a good idea to use GPIO25 on the Stark CMR because of the relay. 
The workaround is easy, just switch the GPIO around to match your needs. I'm not sure how this is best implemented in the project though. We don't want to confuse new users by switching around GPIO "under the hood" without it being very obvious.
I think we need a BE software experts opinion on this.

uitgeslapen, 2025-03-16:
Hello, I think you are absolutely correct, that change would be to drastic and have impact many setups. I think the change should happen in the automatic precharging code/option.  Since it does not work today with the code as is, nobody would be impacted. I think that in that part of the code we should use a different pin " if stark CMR HW use pin x, otherwise pin Y." Optionally we could do this for all users since to my knowledge there is only a couple of other persons that have a working setup.

uitgeslapen, 2025-05-14:
Issue tested and OK

closed 2025-05-14 by uitgeslapen