dalathegreat/Battery-Emulator #984: mqtt add function

jakub8109, 2025-03-18:
Please add a function to mqtt so that I can change the max load and unload parameters from HA.

Max charge speed: 1.0 A 

Max discharge speed: 15.0

jspv33, 2025-08-13:
For those not using HA, enabling mosquitto_pub on discharge and charge current separately would be helpful.

frasersimpson83, 2025-08-20:
I'm keen to be able to control charge and discharge rates from MQTT also from 0 up to 30amp or whatever.   I want to be able to stop charge and discharge completely so going down to zero is important.   

I'm not coder, but anything I can do to help test this etc I am keen.   Thanks

frasersimpson83, 2025-08-20:
My inverter doesn't allow me to control that stuff which is why we're looking to get BE to do it.   There's a few other inverters that others have brought up that have MODBUS control, but it's not implemented very well and has issues and they wanted to have BE control this also to give them other options.    

So yeah in an ideal world the inverter would do it, but it's not possible for everyone.

jspv33, 2025-08-20:
Using MQTT from BE to automate house automation involves some low level coding.
Less than using modbus on the inverter, even less than using http with automated authorization on the inverter.
However the  modbus and http ways will allow other options to control charge and discharge, such as allowing discharge as from a given house usage or time control or SOC setting.
Having other options in BE is to my belief beyond the purpose of BE and difficult to implement as inverters have so many different features.