dalathegreat/Battery-Emulator #990: https://github.com/dalathegreat/Battery-Emulator/pull/939     removed serial

havrla, 2025-03-22:
How to connect paralel I3 battery.  This options require 3x separate CAN interface.

1,lily  native CAN - master battery
1.lily  MCP2515   - slave battery
1.lily serial connect to 2.lily
2.lily  - connect to invertor.

I3 battery has same CAN ID as invertor , not possible using filter.

Please alternate HW or solutions ?

Thanks

lenvm, 2025-03-23:
Hi @havrla!

To use 3 CAN transceivers, you could use ESP32DEVKIT hardware with 3 CAN transceivers:
- SN65HVD230
- MCP2515
- MCP2518FD

havrla, 2025-03-23:
Heloo
thanks @lenvm   Yes, this board is intended for programming experts, I'm not, I'm more familiar with classic electronics. The project currently does not support three CAN interfaces in the user configuration part - that's the part for me, and a few other people who will be using the project.

###########

I’d like to offer a brief analysis regarding the removal of the serial link and the use of two LilyGO units.
Reasons for Removal:

0. Safety
1. Use of a second CAN interface – This is true, but it’s insufficient if the system has two batteries.
2. Possibility to use StarkCMR – Again, only supports two CAN interfaces, which is still a limitation.
3. Use of a CAN filter – This is possibly a good approach, but just like an additional Lily, it means extra hardware. That does not improve safety—on the contrary, maintaining additional software for many different automotive batteries and a wide range of inverters becomes unsustainable and unproductive.

The BMW i3 sends many messages from the vehicle, and the battery sends just as many.
Protocols towards the inverter may be updated over time, making a CAN filter unsustainable.
In the end, a filter is just as risky as a second LilyGO.

In Contrast: Two LilyGO Units = Safer in Parallel Battery Setups
    The first LilyGO reads data from both batteries and reports to the web or MQTT.
    The second LilyGO reads collected data meant for the inverter, and also displays it on the web and MQTT.
This structure makes it very easy to see what data is being sent to the inverter and how it is collected.
Without this capability, it's not possible to effectively debug or develop parallel battery operation.

Due to basic safety checks, installing a sniffing gateway just to see what’s being sent to the inverter is unrealistic for the average DIYer.
Even hobbyists want visibility into what is being sent to the inverter—the more transparency, the safer the system, as one can effectively validate data correctness.
example:
https://github.com/dalathegreat/Battery-Emulator/issues/926

Final Request:

Please retain the serial link for effective use in parallel battery setups.
However, if this is truly a safety issue and the serial link must be removed, I kindly ask for the option to connect two CAN interfaces (e.g., MCP2515/MCP2518, as mentioned above).
This would prevent the need for additional hardware and the complex maintenance of a CAN filter (which is not sustainable, and adds no real safety benefit).
Also, please provide a web display of the data being sent to the inverter, to maintain transparency and control.

Thanks.

lenvm, 2025-03-23:
The DEVKIT hardware is just as much “classic electronics” as the LilyGo, and supports the 3x CAN you are looking for. The LilyGo simply doesn’t have enough free GPIO to support 2x MCP2515 and the additional wake-up pins for the BMW I3.

The project also already supports 3 CAN (for inverter, battery and double battery) via the configuration file.

I therefore think the DEVKIT hardware is still the quickest solution to satisfy your needs.

havrla, 2025-03-23:
@lenvm 
Please, how do I assign the batteries and the inverter to specific interfaces?
/* 
CAN_NATIVE = Native CAN port on the LilyGo & Stark hardware
CANFD_NATIVE = Native CANFD port on the Stark CMR hardware
CAN_ADDON_MCP2515 = Add-on CAN MCP2515 connected to GPIO pins
CANFD_ADDON_MCP2518 = Add-on CAN-FD MCP2518 connected to GPIO pins
*/
volatile CAN_Configuration can_config = {
    .battery = CAN_NATIVE,   // Which CAN is your battery connected to?
    .inverter = CAN_NATIVE,  // Which CAN is your inverter connected to? (No need to configure incase you use RS485)
    .battery_double = CAN_ADDON_MCP2515,  // (OPTIONAL) Which CAN is your second battery connected to?
    .charger = CAN_NATIVE,                // (OPTIONAL) Which CAN is your charger connected to?
    .shunt = CAN_NATIVE                   // (OPTIONAL) Which CAN is your shunt connected to?
};

Is there something like (CAN_NATIVE2,  CAN_NATIVE3) - for DEVKIT ?,  for a third interface?
if yes, then ideal.
secondary ideal - support in lilygo CAN_ADDON2_MCP2515 for secondary external interface and using last pin for CS :-)
Thanks

Please, how can I display the data being sent to the inverter?  for verification  

LilyGO has the option to connect an additional CAN interface — as I mentioned, there is one available pin that can be used for the CS (Chip Select), and the others can be shared.

I’m not looking for a quick fix, but for a high-quality and simple solution.

thanks thanks

lenvm, 2025-03-24:
1. The settings in `USER_SETTINGS.cpp` you can set to the following:

```
volatile CAN_Configuration can_config = {
    .battery = CAN_ADDON_MCP2515,           // Which CAN is your battery connected to?
    .inverter = CAN_NATIVE,                 // Which CAN is your inverter connected to? (No need to configure incase you use RS485)
    .battery_double = CANFD_ADDON_MCP2518,  // (OPTIONAL) Which CAN is your second battery connected to?
    .charger = CAN_NATIVE,                  // (OPTIONAL) Which CAN is your charger connected to?
    .shunt = CAN_NATIVE                     // (OPTIONAL) Which CAN is your shunt connected to?
};
```

2. The `CAN_1_TYPE` etc. don't need to be set. To my understanding these defines are not yet used by the rest of the code.

3. Indeed, to use the two wake up pins, it makes sense to modify `WUP_PIN2 GPIO_NUM_32` to any other pin that is still free on the ESP32DEVKIT hardware.

havrla, 2025-04-06:
My colleague and I came to the conclusion that we will try to develop a HAT with 2 or 3 MCP2515 for the DEVkit and perform the SW integration and project.

BEMSman, 2025-04-06:
You could always use one of these boards, you can then use up to 4 CAN channels

[image]

lenvm, 2025-04-06:
Please ensure you check that the code also works with multiple MCP2515.

closed 2025-11-07 by havrla