dalathegreat/Battery-Emulator #992: [BUG] External BMSRESET results in "Warning" Event

ruffieuxh, 2025-03-22:
**Describe the bug**
I am inducing BMSRESET's via MQTT. Thereby the PAUSE event is recorded as a "Warning" event but related, subsequent events are recorded as "Info" events.

**To Reproduce**
- Issue an external BMSRESET via MQTT
- Check the event page on the BE

**Expected behavior**
All events related to a normal BMSRESET action should be recorded as "Info" events

**Screenshots**

[image]

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: 8.8.1
- Battery used: NISSAN_LEAF
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: yes
- MQTT: yes

dalathegreat, 2025-04-05:
I would argue that since the pause event is cleared (the warning / the LED will only be yellow while the reset is in progress), it is working exactly as intended. Power cannot be used while the battery is rebooting, thus raising a caution event. There is always a risk of the reboot not being successful, so caution should be taken during reboot. 

I do not feel comfortable with downgrading this to a Info event, since users may then accidentally Pause, and not figure out why there is no power transfer possible. Having the screen turn yellow, with a clear PAUSE_BEGIN event is key to making the software understandable.

So, I suggest we close this as won't do

closed 2025-04-05 by dalathegreat

ruffieuxh, 2025-04-05:
Many thanks Dala for the thought. No problem with leaving it as is. 

My thought was more, that such BMS reset - or any pause - needs to be manually or remote induced or is defined in the software. So it is intentionally performed. Hence it should be know why and when this happens and therefore should rather be for information only.