dalathegreat/Battery-Emulator #993: Contactor turns off after few seconds

Gerwaz3, 2025-03-23:
Hello,
I connected the KIA EV6 battery to the Sofar 20KTL inverter, RGB Led is green, no errors in events, and the contactor turns on after starting the program for a few seconds and turns off. The battery is discharged, SOC=2.5%, SOH=100%, U=614V. The battery is connected according to the diagram - jumpers and 12K resistors. The FDCAN INIT ERROR error is shown in the events, but only once after switching on the power supply, after the software restart it is gone. The battery parameters read are consistent with the measurements, the inverter also shows SOC and SOH from the battery, it even turns on charging, but when the contactor releases it stops charging. What could be the reason for the contactor disconnecting?
Best regards

lenvm, 2025-03-24:
Hi @Gerwaz3, as your SOC=2.5%, could it be that any value is outside the values listed below?

https://github.com/dalathegreat/Battery-Emulator/blob/7698b52613ba3ded8830c4ccf81a7353f1621c4a/Software/src/battery/KIA-E-GMP-BATTERY.h#L9-L18

Gerwaz3, 2025-03-25:
Hi, my battery is 180 cells version. Battery readings are within defined limits. Here is log:
    2159.039 Values from battery: 
    2159.039 SOC BMS: 7.0%  |  SOC Display: 2.5%  |  SOH 100.0%
    2159.040 0.0 Amps  |  613.0 Volts  |  0 Watts
    2159.040 Allowed Charge 0 W  |  Allowed Discharge 0 W
    2159.040 MaxCellVolt 3400 mV  No  4  |  MinCellVolt 3380 mV  No  93
    2159.040 TempHi 9ТАC  TempLo 8ТАC  WaterInlet 127ТАC  PowerRelay 0ТАC
    2159.040 Aux12volt: 13.4V  |  
    2159.040 BmsManagementMode 0  |  BmsIgnition OFF  |  PowerRelay OFF  |  Inverter 713 Volts
On pin 7 there is a voltage of about 4.3V, I also tested with a smaller resistor than 12K, but the reading in the log is still 127. The room temperature is 10C, on the cells it is correct, but the contactor temperature is 0C (the thermistor in the contactor has a resistance of 18K). Pin 12 (ignition) is connected to pin 1 and 2 (13,4V) but in log ignition is OFF.
I also saw a leak sensor in the code, but the connector description doesn't say anything about it. I have pin 13 (crash signal) unconnected and it has 4.93V - is this correct?

ilia92, 2025-03-28:
Hello @Gerwaz3,

The development of the right CAN messages, turning on the PRA (relays) are not completely developed and currently the battery has around 30% of it's full capacity - at least - mine is 192cells and goes from 3550mV to 3880mV. Out of this range it switches OFF. When you push current - BMS switches off above this voltage, when pulling current - under this voltage. 
I highly suspect you face this kind of issue.

What you can try - Do not pull current (or set it to 0), instead, then charge the battery to around 3600mv/cell and try again. 
To use the full capacity, you can turn the PRA manually by lilygo (with additional relays), similar like Leaf.

I recommend joining Discord channel, where this topics are discussed.

Gerwaz3, 2025-03-29:
Hello, I turned on the contactor manually, the battery charged to 33% (3620mV/cell) but after connecting the contactor back to the BMS nothing improved. I also changed a few initial values ​​in the KIA-E-GMP-BATTERY.cpp file:
`static uint16_t soc_calculated = 500;`
`static uint16_t SOC_BMS = 500;`
`static uint16_t SOC_Display = 500;`
After these changes, the low battery warning no longer appears after restarting.
` #ifdef KIA_E_GMP_BATTERY
    content += "<h4>Cells: " + String(datalayer_extended.KiaEGMP.total_cell_count) + "S</h4>";
    content += "<h4>12V voltage: " + String(datalayer_extended.KiaEGMP.battery_12V / 10.0, 1) + "</h4>";
    content += "<h4>Waterleakage: " + String(datalayer_extended.KiaEGMP.waterleakageSensor) + "</h4>";
    content += "<h4>Temperature, water inlet: " + String(datalayer_extended.KiaEGMP.temperature_water_inlet) + "</h4>";
    content += "<h4>Temperature, power relay: " + String(datalayer_extended.KiaEGMP.powerRelayTemperature) + "</h4>";
    content += "<h4>Batterymanagement mode: " + String(datalayer_extended.KiaEGMP.batteryManagementMode) + "</h4>";
    content += "<h4>BMS ignition: " + String(datalayer_extended.KiaEGMP.BMS_ign) + "</h4>";
    content += "<h4>Battery relay: " + String(datalayer_extended.KiaEGMP.batteryRelay) + "</h4>";
#endif  //KIA_E_GMP_BATTERY `

Gerwaz3, 2025-04-12:
Hello, I noticed that after switching on the power supply CANMCP2517FD_INIT_FAILURE appears, you can delete the events or restart the program and it will not happen again. The contactor is also switched on for those few seconds only after switching on the power supply, restarting will not switch on the contacts. Can someone tell me which wires from the contactor mean what?
[image]
The thin blue and black are probably the temperature sensor, the thicker yellow one I connected to GND, the blue and gray one I connected to 12V, the green one had no connection to the others.
If anyone uses such a battery, please send me a LOG like the one I have above, I will have it for comparison.
I have a suggestion to add additional information about the battery:

KIA-E-GMP-BATTERY.cpp :
`  // Update webserver datalayer
  datalayer_extended.KiaEGMP.total_cell_count = datalayer.battery.info.number_of_cells;
  datalayer_extended.KiaEGMP.battery_12V = leadAcidBatteryVoltage;
  datalayer_extended.KiaEGMP.waterleakageSensor = waterleakageSensor;
  datalayer_extended.KiaEGMP.temperature_water_inlet = temperature_water_inlet;
  datalayer_extended.KiaEGMP.powerRelayTemperature = powerRelayTemperature;
  datalayer_extended.KiaEGMP.batteryManagementMode = batteryManagementMode;
  datalayer_extended.KiaEGMP.BMS_ign = BMS_ign;
  datalayer_extended.KiaEGMP.batteryRelay = batteryRelay;`

datalayer_extended.h :
`typedef struct {
  uint8_t total_cell_count = 0;
  int16_t battery_12V = 0;
  uint8_t waterleakageSensor = 0;
  int8_t temperature_water_inlet = 0;
  int8_t powerRelayTemperature = 0;
  uint8_t batteryManagementMode = 0;
  uint8_t BMS_ign = 0;
  uint8_t batteryRelay = 0;
} DATALAYER_INFO_KIAEGMP;`

advanced_battery_html.cpp :
`#ifdef KIA_E_GMP_BATTERY
    content += "<h4>Cells: " + String(datalayer_extended.KiaEGMP.total_cell_count) + "S</h4>";
    content += "<h4>12V voltage: " + String(datalayer_extended.KiaEGMP.battery_12V / 10.0, 1) + "</h4>";
    content += "<h4>Waterleakage: " + String(datalayer_extended.KiaEGMP.waterleakageSensor) + "</h4>";
    content +=
        "<h4>Temperature, water inlet: " + String(datalayer_extended.KiaEGMP.temperature_water_inlet) + "</h4>";
    content +=
        "<h4>Temperature, power relay: " + String(datalayer_extended.KiaEGMP.powerRelayTemperature) + "</h4>";
    content += "<h4>Batterymanagement mode: " + String(datalayer_extended.KiaEGMP.batteryManagementMode) + "</h4>";
    content += "<h4>BMS ignition: " + String(datalayer_extended.KiaEGMP.BMS_ign) + "</h4>";
    content += "<h4>Battery relay: " + String(datalayer_extended.KiaEGMP.batteryRelay) + "</h4>";
#endif  //KIA_E_GMP_BATTERY`

`#if !defined(BMW_PHEV_BATTERY) && !defined(BMW_IX_BATTERY) && !defined(BOLT_AMPERA_BATTERY) &&       \
    !defined(TESLA_BATTERY) && !defined(NISSAN_LEAF_BATTERY) && !defined(BMW_I3_BATTERY) &&          \
    !defined(BYD_ATTO_3_BATTERY) && !defined(RENAULT_ZOE_GEN2_BATTERY) && !defined(CELLPOWER_BMS) && \
    !defined(MEB_BATTERY) && !defined(VOLVO_SPA_BATTERY) && !defined(VOLVO_SPA_HYBRID_BATTERY) &&    \
    !defined(KIA_HYUNDAI_64_BATTERY) && !defined(KIA_E_GMP_BATTERY) //Only the listed types have extra info`

Gerwaz3, 2025-05-02:
Hello, I see that there is little interest in this type of battery, so I am closing the topic. I will only add that the problem disappeared after further charging to 90% (more than 3.9V). Now the contactor holds, I discharge the battery to check at what % it will let go.

closed 2025-05-02 by Gerwaz3