dalathegreat/Battery-Emulator #994: [BUG] BMW i3 60, Precharge: Error precharge blocked

alikator, 2025-03-23:
**Describe the bug**
I had LilyGo with CAN connected to the BMW i3 60A and  with 2515 CAN connected to the FoxESS H1 and all was working fine charging and discharging (max 5KWh) until yesterday evening when system switched off by itself. When I found it all devices was Off. I start to give a current to the inverter from local provider and inverter seems fine 100%, LilyGo start to work too, but not the battery. The battery´s relays was silent. LilyGo reports "Precharge: Error precharge blocked" and another strange message in Events is Battery Overheat (but it isn´t because LilyGo shows 17ºC). So I disconnect and reconnect Low voltage connector when 12v was ON, and sounds like initial relays start execute, but not the HV relay to power Invertor.
So now the battery make relay clicks on start and not connect HV relay pass high voltage and stays Precharge: Error precharge blocked.
All cells are fine, Inverter can see the % of the battery and is Waiting for HV load from battery.
I was trying to change PRECHARGE_TIME_MS less and more then 500, but issue was not solved, I was trying to use different Lilygos and result is the same.
I was trying to change different versions of Battery-Emulator and it was not solved an issue.
Contactor control not giving the option to compile claiming to some IO25

**To Reproduce**
There is no way to reproduce because it seems some eventual

**Expected behavior**
Battery must work as before, make clicks of all relays and provide power to inverter.

**Screenshots**
Added

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging - ok
- Software version: 3.1.3

[USER_SETTINGS.zip](https://github.com/user-attachments/files/19408842/USER_SETTINGS.zip)

- Battery used: [BMW_I3_BATTERY]
- Inverter communication protocol: [FOXESS_CAN]
- Hardware used for Battery-Emulator: [HW_LILYGO]
- CONTACTOR_CONTROL: [no]
- MQTT: [no]

[image]
[image]
[image]
[image]

havrla, 2025-03-23:
Your issue originates in the battery. The web interface displays a value that is directly taken from the battery's CAN message:

datalayer_extended.bmwi3.ST_precharge = battery_status_precharge_locked;
battery_status_precharge_locked = (rx_frame.data.u8[1] & 0x0C) >> 2;

So essentially, the battery itself is stating that precharge will not proceed.

The “Overheat” message commonly appears when there’s a loss of communication with the battery.
Is this message still being displayed now?

The root cause might be inside the battery, which brings me to a new idea:
it would be valuable to integrate DTC (Diagnostic Trouble Code) reading from the battery into this project, along with error interpretation.

I’ll try to prepare some groundwork for Dala to implement this functionality.

alikator, 2025-03-27:
Hi! Thank you for additional information.
I suspected it by the behaviour of the battery which was  working fine a a weeks and had no excess on charge/discharge, act that way. 
Just to inform that before this happened, i noticed that invertor had EPS overload execution over 5KW with autorecovery, but it already happened  few times before this day without any issue. But this time something gone wrong way so since exactly that invertor´s EPS overload the battery generated this state.

havrla, 2025-03-27:
Idea, you try lilygo configuration only with battery, without invertor.   I have one lilygo+capacitor for testing battery . :-)

alikator, 2025-03-29:
> Idea, you try lilygo configuration only with battery, without invertor. I have one lilygo+capacitor for testing battery . :-)
Sorry for not mentioned before, but that was exactly the first action what did I do when it failed.
Well. Today also I downloaded files from this source and the latest one is 8.9.0
Then I uncomment #define BMW_I3_BATTERY , check that .battery = CAN_NATIVE and the rest of settings are as-is by default.
Fail persist as before.

alikator, 2025-04-03:
Ok. I found a cable to connect my i3 to this battery and check it with ISTA. The cause of blocking was that 12V power supply registered a low voltage and it trigger errors so the HV battery blocks. So I clear all errors with ISTA few times until all errors has been cleared and battery back to work again. 
**The reason of the issue**: 12V raw battery without output voltage regulation that cause low voltage issue.
**Advice**: use regulated bower supply backup that will shut off on the voltage lower then recommended.
**Suggestion**: include some kind of error clearing functions to the Emulator

alikator, 2025-04-07:
No, the issue just back without a backup 12v battery. It was working fine this morning, then I switched off 12V power and the rest of installation too. And now decided to start again and found that precharge is blocked again.

dalathegreat, 2025-04-09:
Hi @alikator , nice work atleast troubleshooting it. If we can get a CAN log of the ISTA fault code clearing, I can add it to the Battery-Emulator!

alikator, 2025-04-09:
> Hi [@alikator](https://github.com/alikator) , nice work atleast troubleshooting it. If we can get a CAN log of the ISTA fault code clearing, I can add it to the Battery-Emulator!

Actually, it would be great for all of us.
There was generated more issues.
So 7 of april at morning I've checked that battery well charged at 99% and just switched off 12V power connected to Lilygo, so the battery switched off and all looks like well.
At evening I back to installation and try to switch on the 12V power source for Lilygo to start the battery, but it didn't start. So I connected and foun de next errors, attached. I was not able to delete them with ISTA last version.
So now I am completely disappointed, how the hell the contactors could be welded....
I have #define PRECHARGE_TIME_MS 1000  

I have no idea how it could happened only switching off the LiLygo emulator.
Any Idea how to resolve it?

[image]
[image]

dalathegreat, 2025-04-09:
@havrla , we need to capture a CAN log from a DTC clear, so we can add this functionality to the Battery-Emulator. It will be extremely annoying if people need ISTA to run i3 batteries in storage mode!

havrla, 2025-04-10:
i prepare sniff toomorow :-)  delete DTC and read DTC

havrla, 2025-04-12:
[read_delete.log](https://github.com/user-attachments/files/19718625/read_delete.log)
[read-delete2.txt](https://github.com/user-attachments/files/19718623/read-delete2.txt)
[read-delete2.log](https://github.com/user-attachments/files/19718624/read-delete2.log)
[read-delete.txt](https://github.com/user-attachments/files/19718626/read-delete.txt)

havrla, 2025-04-12:
[2xdeleteISTA.log](https://github.com/user-attachments/files/19719194/2xdeleteISTA.log)

alikator, 2025-04-13:
I can see you did a great job, what will be the next step?

> [read_delete.log](https://github.com/user-attachments/files/19718625/read_delete.log) [read-delete2.txt](https://github.com/user-attachments/files/19718623/read-delete2.txt) [...]

dalathegreat, 2025-04-13:
Massive thanks for the log files @havrla 🙌 I will code this into the Battery-Emulator now!

alikator, 2025-04-14:
Sounds fantastic. Colossal thanks for fast response and dedicated time to @havrla  and @dalathegreat  ! I will test this when it will be ready!

alikator, 2025-04-28:
Hello! Any updates since then? May be any beta or alfa?

alikator, 2025-05-22:
May I help you in this case?