dalathegreat/Battery-Emulator #1000: BMS reset pin interferres with dual battery when using 2 nissan leaf batteries

szzer, 2025-03-24:
Dear Developers,

I'm in the process of adding a second battery to my Home storage setup.
The current setup is the following:
Goodwe GW8K-ET + Lillygo + 24KWH nissan leaf battery, 2013 model. 
the wiring diagram i'm using at this moment is the legacy wiring (so without BMS reset pin).

I want to add a second nissan leaf battery, since they are super cheap here in the Netherlands.
AND add the BMS reset function

To add the second battery I read in the Wiki that a second can interface is needed.
https://github.com/dalathegreat/Battery-Emulator/wiki/Double-Battery

The wiring diagram dictates that the following pins need to be used for the can interface:
pin, 12 , 5, 34, 18, and 35 
https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-add%E2%80%90on-(MCP2515)

This leaves the no pin free for the bms reset pin, In the picture it is located on pin 18.

Is it key to have the frequent bms reset ? If so, how enable the BMS reset in combination with a second nissan leaf battery?

lenvm, 2025-03-24:
Hi @szzer,

To support your setup with BMS reset, as the number of pins on the LilyGo are limited, you could consider using one of the alternative hardware with more free GPIO:

- https://github.com/dalathegreat/Battery-Emulator/wiki/Hardware:-ESP32-DevKit-V1
- https://github.com/dalathegreat/Battery-Emulator/wiki/Hardware:-Stark-CMR

Hope this helps!

szzer, 2025-03-25:
Dear Lenvm, 

Thanks for the advise.
I do have a ESP32 devkit v1 lying around. So will try to get that to work.
Unfortunately less well described in the wiki though.

Will keep you updated on the progress.

dalathegreat, 2025-03-25:
Another way to gain additonal pins on the LilyGo board, is to use an SD card breakout board, and use the GPIO pins normally reserved for SD.

This adapter for instance works: https://shop.exploitee.rs/shop/p/microsdemmc-breakout-board

Then just edit the HAL file for lilygo, to map the BMS_RESET pin to one of the freed up pins!

szzer, 2025-04-30:
Just a check, do I need a set of relay contactors for each battery, so in this case 2 times 4 contactors (8 relays in total)? 
Or can I connect both batteries to one set of relayscontactors?

szzer, 2025-04-30:
Another question;

To equalize the voltage between the two batteries, would it make sense to install a precharge resistor between the two batteries?
This would then be controlled by the battery emulator, to equalize the battery voltages, Once battery voltages are equalized the precharge resistor the main contactor closes and te precharge contactor is opened, just like in the battery itself?

This would make the risk of fuse blowout a lot less.

szzer, 2025-04-30:
I get a compilation error I cannot find the cause of:

"
Compilation error: 'BMS_POWER' was not declared in this scope
"

can you help with finding my mistake? 

the settings in user_settings.h I've set are:

`#ifndef __USER_SETTINGS_H__
#define __USER_SETTINGS_H__
#include <WiFi.h>
#include <stdint.h>

/* This file contains all the battery/inverter protocol settings Battery-Emulator software */
/* To switch between batteries/inverters, uncomment a line to enable, comment out to disable. */
/* There are also some options for battery limits and extra functionality */
/* To edit battery specific limits, see also the USER_SETTINGS.cpp file*/

/* Select battery used */
//#define BMW_I3_BATTERY
//#define BMW_IX_BATTERY
//#define BMW_PHEV_BATTERY
//#define BOLT_AMPERA_BATTERY
//#define BYD_ATTO_3_BATTERY
//#define FOXESS_BATTERY
//#define CELLPOWER_BMS
//#define CHADEMO_BATTERY	//NOTE: inherently enables CONTACTOR_CONTROL below
//#define CMFA_EV_BATTERY
//#define IMIEV_CZERO_ION_BATTERY
//#define JAGUAR_IPACE_BATTERY
//#define KIA_E_GMP_BATTERY
//#define KIA_HYUNDAI_64_BATTERY
//#define KIA_HYUNDAI_HYBRID_BATTERY
//#define MEB_BATTERY
//#define MG_5_BATTERY
#define NISSAN_LEAF_BATTERY
//#define ORION_BMS
//#define PYLON_BATTERY
//#define DALY_BMS
//#define RJXZS_BMS
//#define RANGE_ROVER_PHEV_BATTERY
//#define RENAULT_KANGOO_BATTERY
//#define RENAULT_TWIZY_BATTERY
//#define RENAULT_ZOE_GEN1_BATTERY
//#define RENAULT_ZOE_GEN2_BATTERY
//#define SONO_BATTERY
//#define SANTA_FE_PHEV_BATTERY
//#define SIMPBMS_BATTERY
//#define STELLANTIS_ECMP_BATTERY
//#define TESLA_MODEL_3Y_BATTERY
//#define TESLA_MODEL_SX_BATTERY
//#define VOLVO_SPA_BATTERY
//#define VOLVO_SPA_HYBRID_BATTERY
//#define TEST_FAKE_BATTERY
#define DOUBLE_BATTERY  //Enable this line if you use two identical batteries at the same time (requires separate CAN setup)

/* Select inverter communication protocol. See Wiki for which to use with your inverter: https://github.com/dalathegreat/Battery-Emulator/wiki */
//#define AFORE_CAN        //Enable this line to emulate an "Afore battery" over CAN bus
#define BYD_CAN          //Enable this line to emulate a "BYD Battery-Box Premium HVS" over CAN Bus
//#define BYD_CAN_DEYE     //Enable this line to emulate a "BYD Battery-Box Premium HVS" over CAN Bus, with Deye specific fixes
//#define BYD_KOSTAL_RS485 //Enable this line to emulate a "BYD 11kWh HVM battery" over Kostal RS485
//#define BYD_MODBUS       //Enable this line to emulate a "BYD 11kWh HVM battery" over Modbus RTU
//#define FERROAMP_CAN     //Enable this line to emulate a "Pylon 4x96V Force H2" over CAN Bus
//#define FOXESS_CAN       //Enable this line to emulate a "HV2600/ECS4100 battery" over CAN bus
//#define GROWATT_HV_CAN   //Enable this line to emulate a "Growatt High Voltage v1.10 battery" over CAN bus
//#define GROWATT_LV_CAN   //Enable this line to emulate a "48V Growatt Low Voltage battery" over CAN bus
//#define PYLON_LV_CAN     //Enable this line to emulate a "48V Pylontech battery" over CAN bus
//#define PYLON_CAN        //Enable this line to emulate a "High Voltage Pylontech battery" over CAN bus
[... 129 lines cut ...]
extern volatile float CHARGER_END_A;

extern volatile unsigned long long bmsResetTimeOffset;

#ifdef EQUIPMENT_STOP_BUTTON
typedef enum { LATCHING_SWITCH = 0, MOMENTARY_SWITCH = 1 } STOP_BUTTON_BEHAVIOR;
extern volatile STOP_BUTTON_BEHAVIOR equipment_stop_behavior;
#endif

#ifdef WIFICONFIG
extern IPAddress local_IP;
extern IPAddress gateway;
extern IPAddress subnet;
#endif

#if defined(DEBUG_VIA_USB) || defined(DEBUG_VIA_WEB) || defined(LOG_TO_SD)
#define DEBUG_LOG
#endif

#if defined(MEB_BATTERY)
#define PRECHARGE_CONTROL
#endif

#endif  // __USER_SETTINGS_H__
`

szzer, 2025-05-01:
Fortunately I've fiddled with arduino's and esp a bit already, I'm up for the challenge.

szzer, 2025-06-24:
A small update:
I have managed to install the dev board with the settings set to nissan leaf battery and dual battery at the same time. 
It works now with 1 battery, if it works with an second battery attached I will find out in the comming weeks, since the dual battery setup is located at my parents place. 
Unfortunately I was not able to test the dual battery setup with only one battery, connecting one battery to the second battery input did not do anything. 
At least the second can breakout board, MCP2515 board from Seeed Studio (option: 0 in the wiki) does not through an error, after I did some proper reading of the wiki (having the print mirrored threw me off big time).

Is this because the second battery reading only starts once the first battery is detected?

Will keep you posted once I get around finishing the setup with 2 batteries at my parents place.

Another interesting discovery I made was on connectors for the Leaf battery:
there is another part number of Renault that fits the Leaf battery: the 291Y16581R it is rather short but was only 15 euro at my local breakers yard.

dalathegreat, 2025-11-17:
Closing, as HAL was refactored