dalathegreat/Battery-Emulator #1016: Help: Using Battery Emulator with Sofar inverter via RS485 - Modbus error 226

dromihetes, 2025-03-27:
Hi!
I'm working on integrating the Battery Emulator with a Sofar HYD 10KTL inverter using RS485 (Modbus) and a RJXZS BMS. The LilyGO T-CAN485 board is used for communication.

We initially tried CAN protocol but moved to RS485 because the RJXZS BMS CAN speed (250kbps) conflicted with Sofar’s 500kbps.

Setup:

LilyGO T-CAN485 connected to inverter RS485 port (pins 10 and 11 as per Sofar manual)

Emulator running with RS485_INVERTER_SELECTED and SOFAR_MODBUS defined

Modbus set to 9600 baud, address 1

Inverter in standby

Wiring verified and matches schematic

Problem:
We constantly get Modbus error: 226 in serial monitor. Battery data is read correctly from BMS via CAN, but the inverter doesn’t respond on RS485.

Question:
Has anyone managed to successfully communicate with a Sofar inverter via RS485 using Battery Emulator? Any tips or updated Modbus map?

Thanks in advance!

dalathegreat, 2025-04-06:
We do not have Sofar RS485 support.

**Use a separate CAN channel for the Sofar inverter. You have a few options;**

If you have a CAN inverter, you need to connect it to a separate 500kbps CAN channel, since the BMS runs at 250kbps on the native CAN

- One option is to use [add on MCP2515](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-add%E2%80%90on-(MCP2515)) board
- Another options is to use [add on CAN-FD MCP2518](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN%E2%80%90FD-add%E2%80%90on-(MCP2518FD)) board
- Third option is to use [Stark CMR board](https://github.com/dalathegreat/Battery-Emulator/wiki/Hardware:-Stark-CMR)

closed 2025-04-06 by dalathegreat

dromihetes, 2025-04-10:
ok thank you very much!!!!!

În dum., 6 apr. 2025 la 14:24, Daniel Öster ***@***.***> a
scris:

dromihetes, 2025-04-10:
hello
my question is whether the emulator can also transmit data via CAN directly
to the inverter or not, that is, to take it directly from the CAN port and
transmit it to the inverter. I ask this because the CAN port on the BMS is
completely turned off so I have no voltage on it, can it be activated from
the web UI?

EgertKanep, 2025-04-11:
Hey!

Following up this discussion.When I look from Sofar HYD manual, it says it support BMS communication over rs485 as well.
So in theory if we address table we could use LilyGo board with Sofar?

[image]