dalathegreat/Battery-Emulator #1019: Renault Zoe bms

arjank3624, 2025-03-28:
Dag allen, ik heb een Renault Zoe accu van 22Kw, deze heb ik aangesloten op de Lilygo maar geeft aan dat 12 cellen een te lage spanning hebben, dit is niet correct want op de cellen meet ik wel de juiste 3,9 volt, de zekeringen zijn ook goed. Nu heb ik een andere BMS gekocht en deze heeft hetzelfde probleem, iemand enig idee wat dit zou kunnen zijn?

Groeten Arjan

arjank3624, 2025-03-28:
Hello everyone, I have a Renault Zoe battery of 22Kw, I connected it to the Lilygo but it indicates that 12 cells have too low a voltage, this is not correct because I do measure the correct 3.9 volts on the cells, the fuses are also good. Now I bought another BMS and it has the same problem, does anyone have any idea what this could be?

Greetings Arjan

arjank3624, 2025-04-03:
[CAN-Log.docx](https://github.com/user-attachments/files/19588096/CAN-Log.docx)

arjank3624, 2025-04-03:
[image]

arjank3624, 2025-04-03:
[image]

dalathegreat, 2025-04-06:
From the Wiki: If the cell number 48 is low, your internal battery fuse most likely has blown. Then you need to open up the battery and replace the fuse.
https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Renault-Zoe-Gen1#troubleshooting

dalathegreat, 2025-04-22:
Did you get it working?

arjank3624, 2025-04-22:
No, I have a battery-pack without the outside(the box). First i had wires between the 12 accupacks. Now i have the setting as the same as original in the box with bushbars and the fuse in the middle but the same problem. I measure the voltage on the connectors and they all good but the emulator program give another measurement.

Sorry for my bad english.

Greetz Arjan

dalathegreat, 2025-04-23:
A celltap/balancing wire is most likely connected wrong. Good luck with the troubleshooting!

arjank3624, 2025-04-30:
it works, I had the old BMS reconnected, there is a closure in it so there were cell fuses defective again, and the cell wiring checks the voltage across the internal fuse of 275A, I had mounted this fuse on the wrong side of the battery pack, the BMS saw that as a broken fuse

arjank3624, 2025-04-30:
[

[image]

](url)

arjank3624, 2025-04-30:
now i have on de battery emulator a warning: MCP2515 message failed to send. Buffer full or no one on the bus to ACK the message!

i have check the resistance on the CAN and this is 61 ohm
check the crystal mhz
the line #define CAN_ADDON is on

groeten Arjan

dalathegreat, 2025-05-01:
Nice work on identifying the wiring mistake. This is a big reason why we recommend NOT opening the batteries, and instead running them as-is in their automotive battery casing. Much easier to setup and less risks of breaking stuff.