dalathegreat/Battery-Emulator #1021: Add extra GPIO Pins

user63147, 2025-03-30:
Is it possible to add  extra GPIO pins using 
MCP23017 IO Expansion Board, Expands 16 I/O Pins
So can use BMS contractor , Emergency stop button  with solax inverter.

Thanks

dalathegreat, 2025-03-30:
The really cheap fix:
One way to gain additonal pins on the LilyGo board, is to use an SD card breakout board, and use the GPIO pins normally reserved for SD. This adapter for instance works: https://shop.exploitee.rs/shop/p/microsdemmc-breakout-board

Then just edit the HAL file for lilygo, to map the new pins.

The more proper fix:
I recommend the Stark CMR board for demanding hardware setups.

user63147, 2025-03-30:
In the lilygo.hal file

#define SD_MISO_PIN 2
#define SD_MOSI_PIN 15
#define SD_SCLK_PIN 14
#define SD_CS_PIN 13
#define EQUIPMENT_STOP_PIN 35
#define BMS_POWER 18

Don’t  know how to remap the pins.   Do I just change the following values as below?

#define BMS_POWER 13

#define EQUIPMENT_STOP_PIN 14

?

dalathegreat, 2025-03-31:
That is correct.

closed 2025-04-02 by dalathegreat