dalathegreat/Battery-Emulator #1022: BYD ATTO 3 Standard Range Battery Issues

leon413, 2025-03-31:
Cell 105 reads 4095 mV on the cell monitor page.

Cell voltages read correctly on the main page.

Temp sensors 9 and 10 read 215 sensors 1 to 8 read 26 - 27.

Temp sensors read correctly on the main page.

Battery undervoltage fault shows in the events menu.

Contactors close and then open a few seconds later.

- Software version: 8.8.1
- Battery used: BYD ATTO 3 Standard range
- Inverter communication protocol: BYD CAN
- Hardware used for Battery-Emulator: [HW_LILYGO]
- CONTACTOR_CONTROL: [no]
- MQTT: [no]

[BYD ATTO.txt](https://github.com/user-attachments/files/19528224/BYD.ATTO.txt)

[image]
[image]
[image]
[image]
[image]

[canlog_00-10-43.txt](https://github.com/user-attachments/files/19528228/canlog_00-10-43.txt)
[canlog_00-10-08.txt](https://github.com/user-attachments/files/19528229/canlog_00-10-08.txt)

dalathegreat, 2025-04-01:
The code has no support for 50kWh standard range. I can make a PR to add support for this battery later today!

dalathegreat, 2025-04-01:
Fix is now live, direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/atto3-standard-range.zip

PR #1026 (2025-04-13): New Battery: Add support for 50kWh standard range Atto3
### What
This PR implements support for Standard range 50kWh Atto 3 batteries.

### Why
Current implementation does not support 50kWh packs, only extended range 60kWh packs. Fixes #1022 

### How
We now startup in NOT_DETERMINED_YET state, and later on switch between STANDARD_RANGE or EXTENDED_RANGE

Switching between these modes happens automatically, and it will affect the following values:
- Max pack voltage
- Min pack voltage
- Cell count (104 vs 125)
- Estimated SOC from voltage (New map added for standard range
- Total pack capacity [... cut]
closed 2025-04-13 by dalathegreat