dalathegreat/Battery-Emulator #1025: Help: Deye Inverter 20kW cannot communicate with BYD Dolphin battery.

MacThailand, 2025-04-01:
Hi all,

I've been trying to use the battery emulator but the inverter showed up the F58 fault (BMS communication). Another issue is after I disconnected the LilyGO, the inverter is still ON. Have you ever faced this issue or do you have any suggestions on how to fix it?

stormbreaka, 2025-04-03:
What I experienced with my SG01HP3 Deye inverter: every change in battery protocol settings needs at least one shut down procedure of the inverter; furthermore I needed to disconnect the plug of the BMS input at Deye;

dalathegreat, 2025-04-05:
Is this a Standard range Atto3 (50kWh), or the Extended range Atto3 (60kWh)?

We do not support the standard range yet, but this pull request aims to resolve that: https://github.com/dalathegreat/Battery-Emulator/pull/1026