dalathegreat/Battery-Emulator #1035: INTERNAL_OPEN_FAULT ERROR on Tesla Y battery

nizmot, 2025-04-05:
Hello,
I have a problem with my Tesla Y AWD battery and Deye inverter. 
INTERNAL_OPEN_FAULT
ERROR
High voltage cable removed while battery running. Opening contactors!

What could be the problem? 
Thank you

1ni3shi, 2025-04-06:
did you unplug the HV cable, HVIL jumpers or X098 connector while the battery emulator was running?

nizmot, 2025-04-06:
I did not touch anything. As soon as the emulator start I get this error. I don't hear the contactors.

1ni3shi, 2025-04-06:
You might need to go back and check all your wiring on the battery.
By the way what do you have wired up on the battery?

nizmot, 2025-04-06:
Update:
1. Redid the HVIL jumpers and no changes.
2. Pluged also the second motor connector and bum "works" (the LED is green but I dont hear the contactors)
3. New problem only 9V out of the battery showing 355V on emulator :(

dalathegreat, 2025-04-06:
What part number is your battery? 2024 packs might require DIGITAL_HVIL option

nizmot, 2025-04-06:
1700012-99-B

nizmot, 2025-04-06:
Also the emulator shows SOC 45%
and Deye shows SOC 0%

[image]
[image]

nizmot, 2025-04-06:
Update: Sorry guys I f..... up :)

The problem was that the X098 connector on the battery was not tightened enough. Now it works :)

dalathegreat, 2025-04-06:
- 1700012-99-B is a known working pack, others on the Discord have tested it ✅ 
- Deye shows 0% SOC, since your "Allowed discharge power" is 0A. Meaning battery does not allow battery to discharge, only charge is possible

Nice work! We wrote this specifically in the Wiki _"If the 12V supply is too weak, closing contactors wont be possible. The cables also need to be quite thick to avoid voltage drop. 1.5mm² is too small, the cables need to be 2.5mm² at minimum!"_

Enjoy the battery!

closed 2025-04-06 by dalathegreat

1ni3shi, 2025-04-07:
> What part number is your battery? 2024 packs might require DIGITAL_HVIL option

Hi Dala, Digital HVIL did not work for me. As a result I had to modify the CAN signal to face HVIL OKAY.