dalathegreat/Battery-Emulator #1043: Help: Problem with BYD Dolphin standard range battery.

Bestpattanapen, 2025-04-07:
I used the BYD Dolphin battery (standard range), which has a capacity of 44.9 kWh and 104 cells. I believe the only difference between the Atto 3 Standard Range battery is that instead of using 135Ah battery cells, the Atto3 used 150Ah battery cells instead, with the same arrangement and structure.

I used 8.9.0 version with some modification to byd atto3 battery

My experience.

1.Contactors close and reopen a few seconds later. For whatever reason, same configuration, same application version; sometimes it works (contactors stay closed), but the majority of the time it closes and opens.

2.When the contactors are not re-opened, the Deye inverter displays F58 (BMS Communication Error). Lilygo displays the message "inverter does not respond for more than 60 seconds." BMS and Lilygo can communicate entirely fine; I can see SOC, voltage, temperature, voltage deviation, and voltage for each cell on my phone via Lilygo Wifi

3.The third one, I am not sure if this support to take place or not, when I power off Lilygo, the contactor on the battery is not open (although still power on 12V and IGN to the battery pack).

Do you have any suggestion or experience the same 

Thank you

dalathegreat, 2025-04-07:
Hi @Bestpattanapen 

We do not have support for Dolphin yet, you can see how we just added support for 50kWh Atto3 Standard range in this pull request: https://github.com/dalathegreat/Battery-Emulator/pull/1026

Basically we need to do the same with the Dolphin, since it will have a very different voltage range.

Please try this version, it should help since the voltage range is extended on bootup: Direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/atto3-standard-range.zip

To solve the Deye problem, please install a dedicated CAN channel for the inverter. It sounds like it is interfering with the Dolphin messages. You can get this fixed by these methods:

* You can [add an isolated MCP2515 CAN channel](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-add%E2%80%90on-(MCP2515))
* You can [add an isolated MCP2518 CANFD channel, and run it in classic CAN mode](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN%E2%80%90FD-add%E2%80%90on-(MCP2518FD))
* You can use the [Stark CMR](https://github.com/dalathegreat/Battery-Emulator/wiki/Hardware:-Stark-CMR) hardware

Let me know how the newer software works!

Bestpattanapen, 2025-04-07:
I appreciate your rapid response.

Correct me if I'm mistaken, according to my understanding. 

The voltage battery of the BYD Atto3 Standard Range should work OK because the overall voltage of the battery is quite similar (max charge at roughly 354V +-), with the only difference being the Ah of the battery cells. Thus, the overall energy capacity ranges between 44.9 kWh and 50 kWh.

I tried the version that you suggested, and since my battery's overall capacity is only 44.9kWh So I simply changed the overall capacity in BYD-Atto3-Battery.cpp file from 50,000Wh to 44928Wh and uncommented the estimate SOC in BYD-Atto3-Battery.h file.

The result remains the same. The contactors remain closed and reopen a few seconds later.

Am I doing it correctly, or do I need to add or change some more code to overcome this problem?

Thank you

dalathegreat, 2025-04-07:
If emulator status is Green, and no FAULT event is active, then we can't do anything.

Next step is to get a vehicle log from a BYD Dolphin, so that we can see if the CAN messages to close contactors are different. I have instructions here on how to log: https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging#can-logging-via-webserver

And new integration (this is not a fullfledged new integration) https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

Would be nice if you could log the battery as-is now in standalone mode. We have no CAN logs from a BYD Dolphin.

Then we just need the vehicle CAN logs also.

Let me know if you can provide some CAN logs!

Alternatively, you can bypass contactors, like on a locked battery.

Bestpattanapen, 2025-04-07:
The contractor still has a mind of its own. Sometimes it remains closed, other times it closes and opens.

On the Lilygo side, it remains green until the statement "inverter not sending message via CAN for the last 60 seconds." shows up in the event log, regardless of whether the contractor remains closed or closes and reopens a few seconds later. In all circumstances, the lilygo will remain green before turning yellow after the message is displayed.

On the webserver page, the inverter is always present with a check mark indicating whether it is turned on or off. On the other hand, the battery remains a cross mark while there is no 12V electricity to the battery and a check mark when the battery has 12V power. Is this the reason behind the problem?

I just ordered MCP2515 online. It'll arrive later this month. When it arrives, I will test MCP2515 with Lilygo and let you know the results.

While waiting for MCP2515, if you have any ideas or suggestions that I can attempt before MCP2515 arrives, please do so.

Thank you!

and for the Feedback

When I test the version you recommend to dolphin standard range (the one that supports the Atto3 Standard Battery), it displays the warning "Battery exceeds maximum design voltage; discharge battery to prevent damage." My battery is currently at 346V, so I don't think it's over the maximum design voltage, but when I use the new version that support atto3 standard ,the cell number 105 that appears 4.095V has gone (which is excellent, since I have use atto3 extended range to test dolphin standard battery and the cell#105 appear), but the temperature sensors 9 and 10 still indicate 215C.  

PS. Please wait for me to get the Dolphin CAN log for you.

Bestpattanapen, 2025-04-07:
Hi sir, this is the CANlog for the BYD dolphin battery (standard range). However, I had refresh it several times during the record of the log so the timestamp might be a bit confusing. I couldn't manage to record any CANlog using TEST_FAKE_BATTERY so I used the BYD_ATTO3_BATTERY.

[canlog.txt](https://github.com/user-attachments/files/19631289/canlog.txt)

dalathegreat, 2025-04-09:
Hi, Thanks for the CANlog! it looks like the BYD Dolphin messages are the exact same as the BYD Atto 3. So the code should work great with the Dolphin also.

While you are waiting on parts, can you try removing the inverter CAN, and just try to run the battery? See if contactors stay engaged that way?

bingflash2, 2025-08-29:
Hello @Bestpattanapen, I can prepare a beta version of the battery emulator for you to support your battery if you reply

Best regards, Billy