dalathegreat/Battery-Emulator #1051: Sungrow SH10RT limiting charge/discharge battery current day time to 10A

negis02, 2025-04-12:
Hi,
In my setup i have:
-  SH10RT-V112 hybrid inviter,
- 11.25kW solar array connected in one string - open circuit voltage 950V, max MPPT at 840V,
- 75kW battery assembly of 72s4p CALT li-po cells, so my set "safe" working range is 237.6V-295.0V (night time reaching max 8kW of charge or discharge power at 27A current),
- LILYGO with emulating BYD_CAN, for testing i using TEST_FAKE_BATTERY with sets of actual my battery configuration and design voltages. now running BEv8.9.0
- Location: EU :)

I tested setup with force charge-discharge night time - inverter reaching max 30A current, and after warm up it continuously charging-discharging battery around at 27A
But when sun rising inverter starts limiting current to 10A, is no matter is what kind configuration running Forced mode or Self-consumption mode, Export power limit set to 10000;
 
i read issue:  [huawei inverter #15
](https://github.com/dalathegreat/Battery-Emulator/issues/15#issuecomment-1932718008) and get idea about voltages:
since my solar panels string reaching 950V, but my battery set max voltage is 295V is huge different and inverter internal DC-DC converters day time limiting power to avoid of overheating - little similar to Huawei;
I want to ask of Battery emulator and Sungrow users, what solar panels configurations You have and what battery voltage is using?, and do facing similar **battery charge/discharge 10A current limit day time**?

It will be helpful each way i should do:
split solar array in to two strings - lover PV voltage;
reassemble factory welded battery backs to 144s2p and have around 475V - 590V (in this range event no need 30A charge current to reach full 10kW power) i owned battery without BMS so is in consideration what type of DYI BMS to by 72s or 144 :)

it may  be helpful for this issue: [Operation of BYD emulator with SunGrow #55
](https://github.com/dalathegreat/Battery-Emulator/issues/55)

dalathegreat, 2025-04-13:
Hi,
This is not a safe setup. You need a BMS that can talk to the Battery-Emulator, and balance the battery. For your specific setup (72s), there are a few HV BMS options that are compatible with the emulator:

* DIY HV battery:
   * [RJXZS BMS](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-RJXZS-BMS) ✅
   * [Orion BMS](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Orion-BMS) ✅
   * [Simp BMS](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-SimpBMS) ✅

After installing a supported BMS, I also suggest increasing the number of cells, to boost the voltage. Maybe some SH10RT-user can chime in, but usually we use these inverters with the range 300-400VDC.

mohammad2510, 2025-04-21:
i worked in byd atto 3 and sungrow 20k inverter setup 
i got problem with communication the inverter did not  recognizes the battery 
i tried byd can and sungrow can both didnt work 
i tried use rs port in lilygo 
did have an idea about what this problem 
if y can give me how y do the wiring and programing

negis02, 2025-12-27:
Hi, I solved this issue, so my reply may be helpful.

My setup had an 11.25 kW solar array connected as a single string, with an open-circuit voltage of 950 V and maximum MPPT voltage of 840 V.
I split the solar array into two strings to lower the PV voltage.

I rewired the array on the roof and added extra cables to the inverter. After rewiring, the total circuit voltage is now around 400 V, and during sunny days the inverter charges the battery at 30 A.

I also added extra cooling to the inverter radiator. On sunny days, the temperature was reaching more than 52 °C, which caused the inverter to limit the charging current slightly. With an additional cooling fan, the inverter temperature now stays around 49 °C during daytime operation, as well as at night during battery discharge or grid charging.