dalathegreat/Battery-Emulator #1054: Isolation problem, Polstar 2 battery and Deye 40kw

teslauser, 2025-04-13:
I have this setup
Deye Hybrid Inverter 40kw 
Polstar 2 78kw battery.
Stark CMR

Have connected can-bus and all working from battery to Hybrid Inverter. 
Connected hv cabels from battery to hybrid inverter on ”Batteri 1” 16mm2 cabels.

Have connected HVIL out to in by wire.
Pin 47 on BMS to battery chassis grund.
Stark CMR is flashing green. 

And I try to open connectors but get fault ”HV system isolation resistance monitoring status: false

Dont now what I have done wrong(?)
I dont have cover on battery, so battery is open but all is connected.

Any idea whats wrong?

bingflash2, 2025-04-26:
you could try and figure out where p47 (from harness side) goes  (to what module and how that module is its connect to the battery case, since that is the best point move your p47 loop, obviously check connections by gentle sanding incase of corrosion etc.

use a multimeter to check the connection (ohms) between p47 & p48, if it is 0ohms, since your using a polestar and not a volvo there could be a minor difference. you may need to connect P48 to the battery case also.

this information is at your own risk, i do not own this battery, nor have i tested the above method !!

it is just what i would attempt to do, since the 12v battery GND should go to the vehicle body, it seems logical...please read this entire post ...

before doing anything re-read the guide #https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Volvo-XC40---Polestar-2

79 kWh (usable), 82 kWh (gross) Long range is not listed as supported! is this your battery ? have you got the pinout for your battery from polestar incase p48 has moved ?

you need a LOAD on the hv, when you turn on the battery otherwise you can KILL the ability to close the contactors via can,
a very important thing imho, if it was me, i would purchase the hv dc-dc step down converter from the link in the guide, and wire it in parallel to the inverter, some inverters need to be turned on after the battery, thus this way ensures there is always hv load on the battery, when you goto activate, to not lock your contactors, otherwise youll have to manually wire them, which is harder again...

this post is SPECULATION ! use this information at your own risk, but if it works share so others benefit 

if you do not wish to take the risk, id read every post in this thread #https://secondlifestorage.com/index.php?threads/dalas-battery-emulator.12471/page-7 this is the exact moment dala added the xc40 battery, so obviously someone has done it, and might be in that thread.

why not join patreon and ask yourself ? its ~$80usd for a year on the mega tier and ask, user niclashedlund how he did it ?

this post took over an hour for free, hopefully this helps

Cees-JanBerg, 2025-06-21:
did you solve the problem? I can't get the status "HV system isolation resistance monitoring status: False" off my Polestar battery either.
both GND (pin 48) and pin 47 are connected to the chassis

Cees-JanBerg, 2025-06-22:
I solved it, you have to connect the HVIL together from the driveline connector. I have the 4wd where this is only for the connector to the rear axle. If you have a different type it can be multiple connectors. For the wiring diagram see https://www.loopybunny.co.uk/schematics/MY24_Schematic.pdf

closed 2025-10-07 by dalathegreat