dalathegreat/Battery-Emulator #1058: [Feature request] Growatt ARK / APX battery support

Elektrodude, 2025-04-14:
I am looking for a possibily to run Growatt ARK / APX high voltage batteries with the Fronius Gen24 Inverter. 
Let me know how I can help to support the implementation.

Thanks

ErikssonPer, 2025-04-20:
There's 2 different BMS modules used for ARK batteries, and 1 for APX.
Each with it's own protocol.
The one used for ARK batteries coupled to an SPH inverter have successfully been emulated and full documentation for the CAN protocol are avaliable here.

For the others documentation of the protocol would be very helpful.

bingflash2, 2025-04-28:
do you know someone who has your desired battery in operation ?
go and log the communication between the battery and inverter and post it here

Elektrodude, 2025-04-29:
unfortunately I do not know anyone who has this kind of battery... sorry.