dalathegreat/Battery-Emulator #1062: Tesla LFP battery balancing

1ni3shi, 2025-04-15:
Does anyone know why this is happening to my battery pack? Or is this normal?
2nd half of the pack is unbalanced. 
I have tied to manually balance the battery but outcome is the same. 
[image]

frasersimpson83, 2025-04-17:
Your voltages seem quite high compared to my Tesla LFP pack.   If I try and charge my pack more than when it says it's already at 100% (usually around 3350 - 3400 mv when it's been sitting for 30 minutes or so) then I get the same type of issue where the cell deviation just grows.    I don't think they need to be charged that high...... so I think if you just leave them the SOC will recalculate and it should peak around that 3350-3400mv and things should run better.

1ni3shi, 2025-04-17:
Looks like the numbers 3350-3400mv are correct, so my SOC could be inaccurate as it is trying to overcharge the battery.
Do you scale your SOC? and also do you have it charge to 100% often?

I have mine set to scaled 95 to 5%

frasersimpson83, 2025-04-17:
I think you could take two approaches to it really.   If you charged it till you were seeing lets say 3350 - 3400mv and then stopped charging and disconnected the pack overnight, then a lot of the time it will recalculate the SOC, so you might wake up and find it's now at 100% even though before it might only report 90% say.

The second option now (which is only just available in the most recent release) is to discharge it down below what it thinks is 5% and just discharge it till you get down to say 3150mv (mine is sitting at that now at 5% SOC)    You can do that by using a negative number in the scaled SOC so it will discharge below 5% for instance.   Then do the same, once you get it down to 3150mv or close, then just disconnect and leave it overnight and see what it recalculates the SOC to by the morning.   It might take a few go's at it, but I think the discharging down is probably the best option and it should recalculate the bottom percentage down to something more reasonable and then when you charge it back up it should then stop at something more reasonable also.

I don't use scaling on my pack, I go 5% up to 100% without scaling, so real SOC from the pack.   It's worked well for me if the SOC is actually at those reasonable cell voltage numbers.

I don't get enough power to charge to 100% everyday, or also some days if it's charged to full then it doesn't discharge down to 5% overnight so it might go from 100% down to 40% then back up some days, and other days it's down at 5% overnight and only charges to 50% during the day and then goes back down to 5% etc so a mixed bag and I've had no issues as mentioned.

Good luck :)

frasersimpson83, 2025-04-17:
There's a few things there really.   A lot of those comments come from users with NMC chemistry, so LFP is much better able to handle going high and low more so than other chemistry.  

Also, I think a lot of people want to stretch every last bit of life out of the battery, but I don't care if it lasts 10 years, it's more than done what I need and anything else is a win, but some want to aim for 15 years or more but I realistically think that after 10 years we'll be able to buy a new pack for half the price and twice the capacity so why would I want to stretch the life of an old pack.

My pack was nearly new also (2000kms travelled) where are some peoples packs are already 2 or 3 years old.

So yeah, I think that going 10% to 90% would probably always get longer life than 5% to 100% like I do, but I don't think it's going to be like a 25% difference, but in reality who knows........

I think you need to aim to discharge your pack down to 3150mv or 3175mv somewhere like that and then disconnect it and leave it overnight and see what the SOC resets to, then put it back into service and charge it back up to 100% for the first charge after that so it can get some proper info to recalculate the SOC from, doing small charge and discharge is always going to cause issues with it being able to make an accurate calculation.

frasersimpson83, 2025-04-17:
I am sure that charging to 100% isn't as good as charging to 90% even on LFP, but I am just going to take the approach of getting the best bang for my buck back and at the moment I charge 2 x EV's and push back to the grid so that extra 10% helps........  I'm just going to monitor and see how it goes.

It's been in there 18months so far and is reporting 92% SOH so I don't think it's that bad since most packs lose the biggest chunk of their SOH in the first 12 months anyhow.

1ni3shi, 2025-04-17:
So at 9.2% SOC I still have 3250mv in my cells, is that right? I thought it would be lower, but with the sun out now the battery has started charging again. Now I'm confused with my Tesla battery pack. lol

frasersimpson83, 2025-04-17:
No that sounds too high maybe, but not too far off.   As we discussed because it's probably reporting 9.2% when in fact it's probably more like 15 percent or something.   My pack is at 8.2% and it's at 3200mV, doesn't sound like much, but over a pack with 100's of cells it adds up.

I think what you need to do is setup the scaled SOC and make the discharge a negative number, like -5 or something so that when it says it's at 10% it's reporting 15% to the inverter and the inverter will keep discharging.    Just keep an eye on it whilst it does it in case it starts getting too low, but worst case scenario the BE will see a big change and disconnect the battery anyhow so you're not risking anything.    You're just trying to get the battery down low, then we can charge it back up to close to full (maybe set the high end of the scaled SOC in BE to 90%) and then charge it back up and we'll see what SOC it's reporting.    

Then I think you can leave it to cycle a few times and see if the SOC recalculates.   You're going to need to try and find ways to use up the battery to get it down to 5% regularly so that the battery is able to understand what the bottom end is and what the top end is.   I think the issue you're seeing is because you're not going from empty to full enough so the pack is calculating things wrong.

1ni3shi, 2025-04-30:
I was able to get the SOC down to 1.4% and cells were showing 3222/3234mv.
Might need to disconnect the pack and let it rest and see if anything changes, otherwise this battery pack is not your average LFP pack. Or the chances of incorrect reading from the battery emulator which seems unlikely.