dalathegreat/Battery-Emulator #1069: [Feature request] Pass through accurate SOH information to main front end

frasersimpson83, 2025-04-17:
Currently the main screen of BE shows 99% for SOH which I know is a hang up from when we couldn't read / calculate the SOH from the Tesla battery (maybe others, but I only have the Tesla pack).

Now that we are getting the accurate SOH information through in the 'More info' tab, can we look to pass it through to the main screen now so it's not so confusing to look at ?

I know there will need to be some logic to determine if the info in the more info screen is OK to pass through, but from what I've seen it's been pretty accurate now on a range of batteries so should be able to be relied upon.

matvey83, 2026-10-26:
I am currently working on this,  for 3/Y and legacy S/X packs.

matvey83, 2026-10-28:
https://github.com/dalathegreat/Battery-Emulator/pull/3188