dalathegreat/Battery-Emulator #1076: deye 50kw and byd atto3 long range

mohammad2510, 2025-04-21:
i got problem with the remaining charge  , inverter displays the charge percentage  about 0 and about 60 and keeps changing in a random way.
the canbus twiested and isolated did y recommend use 2 lilygo ?
Or is this problem due to something else?

demon1300, 2025-04-24:
Is the battery from a crashed vehicle? If it is, I'd say the estimated state of charge is fluctuating as expected as this is only estimated for a crashed pack and will vary wildly based on loads etc. If it's not crashed, you need to modify the ATTO3 battery .h file and comment out the section about charge estimation. This will use the BMS and will be accurate.

mohammad2510, 2025-04-26:
Thx for the help , I did that and worked well

demon1300, 2025-04-26:
No problems, glad that worked for you!

closed 2025-04-28 by dalathegreat