dalathegreat/Battery-Emulator #1082: RJXZS contactors

dromihetes, 2025-04-22:
Hello, problem with integrating contactors in the code, has anyone encountered this problem before? I use BMS with a Sofar Hyd 10ktl-3ph inverter.
 I read the bms manual, but it seems I'm doing something wrong. 
And one more question, which protocol is better, Pylon or Sofar? 
Thank You!!!

dalathegreat, 2025-04-22:
For RJXZS, contactors are controlled directly by the RJXZS board. User Manual has more wiring info: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-RJXZS-BMS#where-do-i-get-the-user-manual

[image]

For Sofar inverters, only Pylon works, as documented in the Wiki: https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Sofar#compiling-the-software-for-sofar-inverters

CAN configuration:
- Connect RJXZS to the Native CAN port (for 250kbps operation)
- Connect Sofar to another CAN port (for 500kbps operation) Depending on your hardware

dromihetes, 2025-04-22:
I did all the steps but in the battery emulator the battery status is x. Maybe from the top bms app it's a mistake I'm making?

dalathegreat, 2025-04-23:
The Emulator is not controlling the contactors, so X status means nothing.

dromihetes, 2025-04-23:
ok maybe I didn't explain it properly so the contactors can be operated from the top bms app open and close works but the battery status is x and in the explanations if I remember correctly the webserver was like I know that if the battery status is x then the inverter goes into fault and I have exactly this problem

dalathegreat, 2025-04-23:
Ok you are not explaining the problem very well. You wrote initially:

`Hello, problem with integrating contactors in the code`

But now you say inverter goes into fault.

So best is to check that you are doing everything that the inverter requires:

- Voltage present at inverter input
- Pylon protocol, configured like Wiki explains
- CAN communication working with inverter (on its own separate CAN channel)

dalathegreat, 2025-04-23:
We have multiple people with working Sofar KTL inverters, so just follow the Wiki and double check that you have all the things needed:

- High Voltage present at inverter input
- Pylon protocol, configured like Wiki explains
   - Send1 commented in, (Send0 removed)
   - #define INVERT_LOW_HIGH_BYTES and #define SET_30K_OFFSET set 
- CAN communication working with inverter like Wiki explains (on its own separate CAN channel)
   - inverter configured to  .inverter = CAN_ADDON_MCP2515,

Check the Battery-Emulator Webserver, and see if any Events are active.

dromihetes, 2025-04-23:
[image]

dalathegreat, 2025-04-23:
No Inverter/Battery events are active. Problem must be in wiring or something with inverter. Inspect the error codes on the inverter side, and follow the inverter manual for troubleshooting.

dromihetes, 2025-04-24:
ID157
BMSCommunica
tonFault
Communication failure of lithium battery
Make sure your battery is compatible with
the inverter.
 CAN communication is recommended. Check the communication line or port of the battery and inverter for faults

dalathegreat, 2025-04-24:
Now you almost have it!

- Check termination resistance of CAN bus between Emulator-Inverter
- Check that PYLON is configured properly
- Update software on inverter to latest version

dromihetes, 2025-04-24:
60 ohm  , pylon configured ok (in my opinion), the firmware on invertor is v113562.

dalathegreat, 2025-04-24:
You are extremely close, I unfortunately have no more ideas what to check (I have never used Sofar inverter), but once you do find the issue, we can add this error code "ID157" and how to solve it to the Sofar Wiki page.

Good luck with the last bit! 🤞

ErikssonPer, 2025-04-24:
Could you perhaps make a short can log using the canlogger function and upload here?

ErikssonPer, 2025-04-25:
The logs isn't showing any candata coming from the inverter, I think you need to double heck the wiring between inverter and emulator again.

Morris3190, 2025-06-14:
> I apologize for explaining it in my own way and not in the way it should be explained to a specialist. Yes, I use the pylon protocol, the only problem I encountered with the protocol is that the maximum supported is 255ah per cell and my cell is 314ah. And yes, I added the inverter to an addon can mcp 2515. Once again, I apologize!!!

In the latest release the capacity limitation bug has been fixed.

dromihetes, 2025-06-18:
Yes. I made my own code from 0. Thank You anyway!