dalathegreat/Battery-Emulator #1087: RJXZS BMS on Aliexpress item no longer available

sehrkreativername, 2025-04-24:
The linked product on this [wiki page](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-RJXZS-BMS) is no longer available. I also tried to contact RJXZS directly, no response so far.
Can someone verify, if those similar products i have found are compatible or in fact the same BMS and not dubious? 

- https://www.aliexpress.com/item/4001138179223.html
- https://www.aliexpress.com/item/1005006624811993.html
- https://www.aliexpress.com/item/1005008412534717.html
- https://www.aliexpress.com/item/1005007491715620.html

ErikssonPer, 2025-04-27:
The TOPBMS / CLEVERBMS product was the same, I can't tell if the other variants are though.

I have a  kit from TOPBMS on aliexpress  for up to 120S that I used for a short while but then decided to go with EMUS G1 instead.
I had it powered up for 2 weeks.

It's complete with main bms and display board, current sensor, 10 bms slave boards, tempsensors, charge relay, prechagre relay, precharge resistor, diode, and wiringbut  missing a discharge power relay since I kept that for the emus bms.
A relay can be purchased separately from aliexpress and such.

Kit is for sale if anyone needs.

sehrkreativername, 2025-04-27:
Sure, sounds good, if it works with the emulator i'll take it. 
Do you have the link to this product? 

Thank you in advance!

ErikssonPer, 2025-05-03:
> Sure, sounds good, if it works with the emulator i'll take it. Do you have the link to this product?
> 
> Thank you in advance!

Where are you located?

Yes it worked with the emulator, just had to connect with bluetooth app and make the settings for the batterypack first and then the emulator could talk to it.

JorgenSeemann, 2025-11-04:
> Looked also on Aliexpress, but finally bought the kit on Alibaba. Direct chat communication with vendor and shipped DDP to Europe in about 2 weeks. Not going to promote any vendor but search for '192s BMS' or '96s BMS' in the app depending on the size of the battery and the RJZXS hardware will pop-up.

Might be me being hopeless - but I don't seems to be able to locate *any* RJXZS on alibaba. Maybe it's all rebranded topbms ?

Morris3190, 2025-11-05:
Just tested search criteria 'bms 4s 192s 200a' and the RJXZS appears in the top 5 results. Check also for bms_yl96s200a or bms_yl192s200a or typos like bws_yl* in de description.
Shenzhen Chuangheying Electronic Technology (CHY) is a reputable source.