dalathegreat/Battery-Emulator #1103: Charging Battery with the On Board AC Charger

Eletric1, 2025-05-02:
Hi, i am running a Tesla LFP Battery with the Lilygo and a Gen24. Everything is fine. Now there is the idea to charge in case of a Black Out, the Battery with an emergency generator. This Generator ist connectet to the Battery over the original AC Harness to the internal AC On Board Charger. What do you mean, is it possible to do that? If it would be possible, that could be a very nice way to charge the battery in case of emergency!!

bingflash2, 2025-05-04:
yes, this is possible but not with the tesla PCS, because the PCS is expecting communications from a CCS2/NACS vehicle charger.

read this link #https://github.com/dalathegreat/Battery-Emulator/wiki/Charger:-Chevrolet-Volt-Gen1

a lillygo programmed up to instruct the chevy onboard charger to charge your battery using a generator is possible,

regards, billy

bingflash2, 2025-05-04:
yes, the chevy charger in its factory application, takes AC in via a charging cord, and converts it to hvdc with a max power of 3.3kw,
to charge the hv battery on the car.

the voltage and current are configured via can, thanks to dala, you can program a lillgo, to use the chevy charger to charge any supported hv battery using ac from the wall or in a grid down scenario, using a generator

these settings can be set in the battery emulator code.

nagyrobi, 2025-05-04:
But the way it is supported is to be used standalone, just the battery and the chevy charger connected to each other via HV and a dedicated lilygo, right? Not stated in the docs but I guess you can't connect eg. a Gen24 and a chevy charger simultaneously in the same setup...?

bingflash2, 2025-05-04:
a good electrician could configure this several ways.

warning !! **you will likely damage the fronius battery charger supplying hvdc to it, when its operating.**

manual:
as you mentioned, turn off gen24 lillygo can & hvdc breaker to gen24 

auto :
the AC powering the chevy charger could also open a contactor supplying power to the 'gen24 lillygo',
 the lack of commmunications stops the fronius load. a delay powers up a hvdc contactor breaking the hvdc connection to the gen24 (isolating it), a second delay powers the 'chevy lillygo') and backup AC charging via chevy charger begins..

via solar on gen24:
the chevy charger is only capable of 3300w/8a (programmable), you could turn off one of your solar arrays and wire it to the chevy charger output, gen24 sees 'solar' and off you go, **this is at your own risk**, as capacitors inside the chevy unit could be capable of short bursts of current greater than the gen24 solar Imax, 

tips:
chevy chargers get hot and require water cooling..

you could use 2 chevy chargers to achieve 6.6kw of charging if your generator is capable

you could change the CAN message on the chevy charger to 4/5A instead of 8A to accommodate small 2Ka generators like a honda

any AC -> hvdc power supply could do the above job instead of the chevy charger at a slower rate, unless its a kickass power supply, but they do exist

[(https://openinverter.org/wiki/Chevrolet_Volt_Charger)}

Eletric1, 2025-05-05:
Okay, so there is no way to get the PCS AC->DC Converter charging the battery with an plugged in Typ2 cable and an EV Charger Wallbox, because there is no vehicle computer in the background setting MAX SOC and Power!? Is this right? And: Can i CCS charging the Battery because it is directly communicating to the BMS? 

kind regards, Patrick

bingflash2, 2025-05-05:
> Okay, so there is no way to get the PCS AC->DC Converter charging the battery with an plugged in Typ2 cable and an EV Charger Wallbox, because there is no vehicle computer in the background setting [...]

short answer I dont know, Im not a tesla expert. seems like ac charging might work with the right cable (3ph ac charging) as i believe the tesla PCS is a 3 phase charger unlike the BYD model my battery is from which only supports 1xAC+fastDC

ask someone doing tesla EV conversions, as some use the PCS

matvey83, 2026-10-19:
Interesting resource you posted: https://github.com/damienmaguire/Tesla-Model-3-Charger

Thanks

To answer your question - the main contactors must be closed in order for the PCS to charge the pack.   So all HV would be live.