dalathegreat/Battery-Emulator #1105: [Feature request] Add support for Gen1 Zoe 41Kwh double battery

hedkvist, 2025-05-02:
This is actually 2 requests. 
The Secound one with contactors is just a bonus if it's possible. 

Request 1:
Support for double ZOE batteries
I have one ZOE 41Kwh battery running now.
I would like to add another one. 
HW: LilyGo + MCP2515 
Inverter: GoodWe ET 
Battery: 2xRenault Zoe 41kWh 

LilyGo can: Inverter + battery 1. 
MCP2515 can: Battery 2.

Request 2:
I'm using "Contactor_Control" for the single battery right now:
Pre-charge: IO25 
Main Contator: IO32

A function something like this would be awsome:
Double Battery Control ZOE:

IO25 = Precharge Battery1
IO32 = Main Contactor Battery 1
IO33 = Main Contactor Battery 2

1. IF Battery1 and inverter ready for automatic contactor control. 
2. IO32; Main Contator ON, IO25; on for 500ms (Precharge)
3. IF Battery2 Voltage - Battery 1 Voltage = < 2.0V & > -2.0V 
4. IO33: Main Contactor ON

dalathegreat, 2025-05-02:
I was gonna whip this up, but the ZoeGen1 has been converter to the OO method, which does not support double-battery easily.

`void RenaultZoeGen1Battery::handle_incoming_can_frame(CAN_frame rx_frame) {`

@kyberias , shall we revert this, or should we start thinking how to handle a second pack on the common image?

kyberias, 2025-05-03:
@dalathegreat I suggest we figure out how to handle double battery with the new base class approach. I was gonna do that soon anyway. In fact, there was an implementation of that in the common image experiment, we need to validate if it was correct.

hedkvist, 2025-05-15:
@dalathegreat And @kyberias I got 1 extra batteri in my garage, so I'm willing to test if you can come up with some code for it hehe :-)

dalathegreat, 2025-06-12:
I have now added support for running 2x Zoe Gen1 batteries at once in PR https://github.com/dalathegreat/Battery-Emulator/pull/1231

Testers welcome :)

WJDijkstra, 2025-06-12:
Too soon, haven't gone out and bought them yet!!!
Well done once again :)

PR #1231 (2025-06-16): Feature: Add support for Double ZoeGen1 batteries :battery: :battery: 
### What
This PR implements support for running 2x Zoe Gen1 batteries at once

### Why
User requested feature, fixes #1105

### How
We use the same method as on Nissan LEAF doubling!
closed 2025-06-16 by dalathegreat