dalathegreat/Battery-Emulator #1113: [BUG]

Jackiestuxedo, 2025-05-06:
**Describe the bug**
No Refresh of SOC and Remining KwH

**To Reproduce**
Simply use battery emulator, Cell Voltage change, also temp and everything else. 
Nevertheless, SOC and Remining KwH don't move, only after BatteryEmulator restart or Hard Restart.
BMS restart do not change value

**Expected behavior**
It should change SOC and Remining Kwh every CAN refresh.

**Screenshots**
Not need

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: 8.1.1
- Battery used: Kangoo 22kwh
- Inverter communication protocol: CAN Solax X3
- Hardware used for Battery-Emulator: [HW_LILYGO]
- CONTACTOR_CONTROL: Yes
- MQTT: Yes

I Already Add : 
  datalayer.system.status.battery_allows_contactor_closing = true;
  datalayer.battery.info.number_of_cells = 96;

That are missing ont Renault-kangoo-battery.cpp.

dalathegreat, 2025-05-06:
Hi,
Thanks for the bug report. The Kangoo 22kWh pack still has some issues, as noted in the RENAULT-KANGOO-BATTERY.cpp file

```
- Values still need fixing
  - SOC% is not valid on all packs
    -Try to use the 7BB value?
  - Max charge power is 0W on some packs
  - SOH% is too high on some packs
  - Add all cellvoltages from https://github.com/jamiejones85/Kangoo36_canDecode/tree/main
```

It looks like you have a pack that is not updating SOC% correctly. We need to switch to using the polled 0x7bb value instead.

Oh, and fixing the contactor closing is implemented in PR https://github.com/dalathegreat/Battery-Emulator/pull/1112 , thanks for the report!

Jackiestuxedo, 2025-05-06:
Ok I will try to work on It. I will post when i avance.

Jackiestuxedo, 2025-05-07:
I already have implemented Cell voltage. But on 7BB, i don't find SOC.
Also we need hard fix Discharge.
But when Cell voltage are < 3.7 the SOC is working on the first ID. 

I'll comeback when it works

Jackiestuxedo, 2025-05-08:
I've tried to make a emulation of SOC based of voltage.
Nevertheless, I can see the same matter.

The calculation is done at the startup (or hard boot/ or after OTA), and after that there is no refresh.

Did you see same thing on other battery brand?

Best regards

dalathegreat, 2025-08-14:
Hi @Jackiestuxedo , did you make any progress on this integration?

dalathegreat, 2026-03-05:
Fixed with the introduction of https://github.com/dalathegreat/Battery-Emulator/pull/2104