dalathegreat/Battery-Emulator #1118: Question for RPI BatteryEmulator

Elektrodude, 2025-05-08:
I hope it's fine if I post a question for the raspberry pi battery emulator version here as I did not find a forum in the respective gitlab project (https://gitlab.com/pelle8/batteryemulator):

I am interestes in using the Batrium BMS with a Fronius Gen24:
There is an external contactor option for this BMS but I do not know if that is needed and if it makes sense at all in my configuration? 
e.g. maybe the Fronius Gen24 has an internal contactor?

Any ideas?

dalathegreat, 2025-05-08:
Fronius Gen24 has an internal precharge and contactor setup. You can apply high voltage directly to it without any form of precharge.

Elektrodude, 2025-05-08:
ok, thanks. and is this contactor also used in case of emergency stop and standby? 
or in which cases does it open?

dalathegreat, 2025-05-08:
You need to ask Fronius this info, it is not available in their manual. We only know that Fronius will open the contactors incase the HV has a leak to protective earth, we have no clue about the other conditions that the Fronius checks.

dalathegreat, 2025-05-08:
Also, we have Batrium BMS support available in this PR, incase you want to get onto the latest version. The RPI version has not gotten any development for over a year, so many mandatory fixes are not available.

https://github.com/dalathegreat/Battery-Emulator/pull/889

dalathegreat, 2025-05-09:
1. External contactors are always a good idea. I would use them
2. External contactors have no impact here, will work the same with and without them. If you want to disable the battery, there are multiple software ways to do so without needing to physically remove voltage
3. Feel free to backport fixes from here to the RPI repo! 
4. External current sensors needed for almost all inverter protocols. Usually not a problem, since every single automotive EV battery contains a current sensor, so we just use that!

kyberias, 2025-05-09:
I think would be somewhat trivial to provide a "PC-build" of the battery emulator for Linux (RPi) and other OS. Whether that makes practical sense for all the batteries and inverters is a different topic. In fact we probably need to do something like that to run code in a unit-testing harness.

dalathegreat, 2025-05-09:
For actually running the code towards inverters/batteries, PC (Linux) build makes no sense at all, this is an embedded real time system with hard timing requirements.

But for unit testing that would be great!

dalathegreat, 2025-05-13:
1. Fronius Gen24 has an internal precharge and contactor setup. You can apply high voltage directly to it without any form of precharge. https://github.com/dalathegreat/Battery-Emulator/issues/1118#issuecomment-2863675866 This means you can NEVER damage a Gen24 with inrush.
2. Depends on the current flow. Automotive contactors are usually rated for 100s of amps, so by using it in a stationary use with just a few Amps, welding the contactors is unlikely. We have so far never seen anyone damage anything when contactors open, and we have hundreds of users.

dalathegreat, 2025-05-14:
I will be closing this issue, since it is not an issue related to the Battery-Emulator.

For further discussion, I suggest;

- Discussions page on this Github repo
- Discussion forums like secondlifestorage.com 
- The Discord server, for live chatting

Good luck!

closed 2025-05-14 by dalathegreat