dalathegreat/Battery-Emulator #1131: [BUG] Hardware crash due to triggered watchdog ACAN2515Handler - MCP2515

lenvm, 2025-05-11:
**Describe the bug**
Setup: 
- ESP32DevKit
- MCP2515 used for inverter CAN bus
- Inverter configured: `SMA_BYD_HVS_CAN`
- Code: latest main (`f764fb0`)

When sending the following CAN message, using other CAN hardware, on the CAN bus connected to the MCP2515
```
(0.01) TX0 5E7 [8] 00 53 4D 41 00 00 00 00
```

the ESP32 with MCP2515 responds with a crash (related to the ACANn2515Handler):
```
00:00:32.346 >        1.363 Wi-Fi Got IP. IP address: xxx.xxx.x.xx
00:01:11.423 > E (40808) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
00:01:11.429 > E (40808) task_wdt:  - core_loop (CPU 1)
00:01:11.432 > E (40808) task_wdt: Tasks currently running:
00:01:11.438 > E (40808) task_wdt: CPU 0: IDLE0
00:01:11.440 > E (40808) task_wdt: CPU 1: ACAN2515Handler
00:01:11.443 > E (40808) task_wdt: Aborting.
00:01:11.446 > E (40808) task_wdt: Print CPU 1 backtrace
00:01:11.451 > 
00:01:11.452 > 
00:01:11.452 > 
00:01:11.452 > 
00:01:11.452 > Backtrace: 0x4000bfed:0x3ffbd220 0x4008f306:0x3ffbd230 0x400f4f96:0x3ffbd250 0x400f0745:0x3ffbd280 0x400f077d:0x3ffbd2b0 0x400ea56c:0x3ffbd2d0 0x400eaa36:0x3ffbd2f0 0x4008f096:0x3ffbd310
00:01:11.468 > 
00:01:11.468 > 
00:01:11.468 > 
00:01:11.468 > 
00:01:11.468 > ELF file SHA256: 0ef765533
00:01:11.471 > 
00:01:11.692 > Rebooting...
00:01:11.692 > ets Jul 29 2019 12:21:46
00:01:11.695 > 
00:01:11.695 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:01:11.700 > configsip: 0, SPIWP:0xee
00:01:11.700 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:01:11.708 > mode:DIO, clock div:2
00:01:11.708 > load:0x3fff0030,len:4660
00:01:11.711 > load:0x40078000,len:15568
00:01:11.714 > ho 0 tail 12 room 4
00:01:11.717 > load:0x40080400,len:4
00:01:11.717 > load:0x40080404,len:3152
00:01:11.719 > entry 0x400805a0
00:01:12.256 > __ OK __
00:01:12.256 >        0.074 Battery emulator 8.13.dev build
00:01:12.265 > [    83][E][Preferences.cpp:483] getString(): nvs_get_str len fail: SSID NOT_FOUND
00:01:12.271 > [    91][E][Preferences.cpp:483] getString(): nvs_get_str len fail: PASSWORD NOT_FOUND
00:01:12.282 >        0.100 Dual CAN Bus (ESP32+MCP2515) selected
00:01:12.286 >        0.105 Can ok
00:01:12.289 >        0.106 CAN FD add-on (ESP32+MCP2517) selected
00:01:12.291 > [   106][E][esp32-hal-cpu.c:123] addApbChangeCallback(): duplicate func=0x400f1e88 arg=0x3ffbe65c
00:01:12.324 >        0.142 Bit Rate prescaler: 1
00:01:12.327 >        0.142 Arbitration Phase segment 1: 63 segment 2: 16 SJW: 16
00:01:12.332 >        0.143 Actual Arbitration Bit Rate: 500000 bit/s (Exact:yes)
00:01:12.338 >        0.153 Arbitration Sample point: 80%
00:01:12.341 >        0.153 Event: The board was reset due to a task watchdog timeout. Inform developers!
00:01:12.396 >        0.214 Creating Access Point: Battery Emulator
00:01:12.399 >        0.215 With password: 123456789
00:01:12.402 >        0.215 Access Point created.
00:01:12.407 >        0.215 IP address: xxx.xxx.x.x
00:01:12.410 >        0.226 Connecting to Wi-Fi...
00:01:12.492 >        0.310 Event: Wifi connected.
00:01:12.495 >        0.311 Wi-Fi connected. RSSI: 70 dBm, IP address: 0.0.0.0
00:01:13.182 >        1.000 Event: Battery exceeding maximum design voltage. Discharge battery to prevent damage!
00:01:13.550 >        1.368 Wi-Fi Got IP. IP address: xxx.xxx.x.xx
```

**To Reproduce**
- Configure `.inverter` to use `CAN_ADDON_MCP2515`, and compile the code using platformio `build_flags = -I include -D HW_DEVKIT -D SMA_BYD_HVS_CAN -D BMW_IX_BATTERY -D CANFD_ADDON -D CAN_ADDON -D DEBUG_VIA_USB -D DEBUG_VIA_WEB`
- Send `(0.01) TX0 5E7 [8] 00 53 4D 41 00 00 00 00` onto the CAN bus that is configured to be used with the inverter

**Expected behavior**
I would expect the code to work just as how the code works when `CAN_NATIVE` is used as hardware, and configured via `.inverter = CAN_NATIVE`.

**Version and settings (please complete the following information):**
- Software version: `f764fb0`
- Battery used: `BMW_IX_BATTERY`
- Inverter communication protocol: `SMA_BYD_HVS_CAN `
- Hardware used for Battery-Emulator: `HW_DEVKIT`
- CONTACTOR_CONTROL: no
- MQTT: no

dalathegreat, 2025-05-12:
Using 
.inverter as CAN_ADDON_MCP2515
and the .battery as CAN_ADDON_MCP2517

**is not supported in any software yet**. If you made the MCP chips share the same pins, the libraries most likely will crash.

lenvm, 2025-05-12:
The MCP2515 and MCP2518 use a completely different set of pins

dalathegreat, 2025-05-13:
@lenvm , can you try doubling the stack size for acan2515? https://github.com/pierremolinaro/acan2515/pull/50/files

lenvm, 2025-05-13:
I’ll give that a try!

lenvm, 2025-05-23:
I tested the code with the only difference of doubling the stack size to `1024 * 2`. The same error occurred.

dalathegreat, 2025-05-23:
Issue has been identified, and fix available in #1156