dalathegreat/Battery-Emulator #1149: Lilygo CAN channel issue with Fake Battery and Solax

BEMSman, 2025-05-17:
I am trying to use the HW_LILYGO with Fake battery and Solax protocol, however no CAN data is output on the onboard CAN interface.
I have tested this on three different boards, same result, I then changed the inverter from Solax to pylon CAN an it works.

Would there be another change I need to make the onboard CAN communicate to the Solax when using Fake Battery?

dalathegreat, 2025-05-20:
Solax has no periodic CAN sending. It only reacts on incoming messages.

[image]

Many inverters also dont use the CAN unless you have high voltage present on the terminals. So most likely Solax cannot be tested with a fake battery, you need the real deal!

closed 2025-05-20 by dalathegreat

BEMSman, 2025-05-20:
Ok, so its by design, would it be easy for me to make an adjustment so the values are pushed out like the Pylon CAN ?