dalathegreat/Battery-Emulator #1150: Path forward for u supported systems

Mikael-Lagerman, 2025-05-18:
I have a Huawei Sun2000 inverter with two strings of 10 respektive 9 panels and a 10 kWh battery.
I like to add an EV battery to this.
What is the community recommendation for integration/interaction with such unsupported system?
Add a second inverter, spoof additional battery module to the luna2000 battery, or other ideas?

ErikssonPer, 2025-05-18:
Start by looking for a copy of the communication protocol used between battery and inverter, if that can't be found you have to make some logfiles of the communication during different operating scenarios like, charging, discharging, idle.

dalathegreat, 2025-05-20:
I suggest adding another inverter that only charges the battery from AC power. That is what most have done in situations like this. For instance a Fronius inverter can charge from AC inputs in the system.

Alternatively, sell the Huawei and buy one that is on the supported list.

I will close this, since it is not a Battery-Emulator issue. More suitable for the Discussions page, or the Discord server!

closed 2025-05-20 by dalathegreat