dalathegreat/Battery-Emulator #1158: [BUG]Cellmonitor not populating, various errors prbbly related.

JeppeDegnbol, 2025-05-21:
Love this project, sorry if it is something stupid, have searched for hours here and elsewhere.
If someone finds info, or knows possible place to look, please dont hesitate to point me to where to search myself.

**Describe the bug**
The cellmonitor is not showing any values even efter days of running, must be something stupid.
I allso get lots of: BATTERY_TEMP_DEVIATION_HIGH, CELL_UNDER_VOLTAGE, CELL_CRITICAL_UNDER_VOLTAGE
ending in FAULT and contactors opening.

Errors come after 20min running, keeps coming, solved by reboot.

The webserver reports values to be correct. allmost no temp deviation or cell delta, was discharing at only 80w.

**Expected behavior**
Was expecting to see cell voltages.
Dont expect temp or delta voltage in AC room allmost no load.
**Screenshots**
screenshot and log from can added.

**Version and settings (please complete the following information):**
- - Software version: 8.12.0
- Battery used: [Zoe Gen2 52kwh] 9000km. no NVROL done. Charges and discharges as asked. 
- Inverter communication protocol: [BYD_MODBUS] Fronius G24 10.0 row 1.35.4-1
- Hardware used for Battery-Emulator: [HW_STARK]
- CONTACTOR_CONTROL: [yes]
Using 6A contactors ssd DIN, DIN PSU adjustable set to 12,6 to clear low voltage 11,5v in webserver and error code.

[image]

[image]

[image]

[image]

[image]

[canlog_05-01-08.txt](https://github.com/user-attachments/files/20371460/canlog_05-01-08.txt)
[canlog_05-00-56.txt](https://github.com/user-attachments/files/20371464/canlog_05-00-56.txt)
[canlog_05-00-51.txt](https://github.com/user-attachments/files/20371461/canlog_05-00-51.txt)
[canlog_05-00-36.txt](https://github.com/user-attachments/files/20371462/canlog_05-00-36.txt)
[canlog_05-00-26.txt](https://github.com/user-attachments/files/20371463/canlog_05-00-26.txt)

[USER_SETTINGS.txt](https://github.com/user-attachments/files/20371469/USER_SETTINGS.txt)

dalathegreat, 2025-05-21:
Hi,
This is typically happening when CAN corruption occurs. Can you check that CAN termination is OK, and that you are using a shielded CAN cable that has the shield connected to protective earth in one end?

dalathegreat, 2025-05-21:
I found some severe bugs in the Zoe Gen2 code since the last refactoring for NVROL reset

[image]

I will put up a pull request with some fixes shortly

dalathegreat, 2025-05-21:
@JeppeDegnbol Can you try this PR? https://github.com/dalathegreat/Battery-Emulator/pull/1159

Direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/zoe-gen2-bugs.zip

JeppeDegnbol, 2025-05-22:
Thanks for the swift update. The bugfix worked. The battery has now run for 3hours light load, and 20min max charge with no errors. Awesome. Will report again tomorrow letting it run overnight.

I allso added termination on the STARK on CAN, and replaced the cable for CAN with seperate with shield terminated to earth from 220v supply. No change still no data in Cellmonitor. Is this to be expected, can I help or where to look to try and resolve this ?

:)
As a side comment the webserver buttons to open and close contactor works to open it but not to close it. Only way to get battery going again is reboot.

dalathegreat, 2025-05-22:
Hi @JeppeDegnbol ,
I have attempted to add cellvoltage reading for all 96 cells. After startup, it takes ~30 seconds and all cells should have been read.

Can you test it? :)

Direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/zoe-gen2-bugs.zip

Oh, and if it does not work, please provide a CAN log and I can try to correct any CAN bitshifting mistakes!

JeppeDegnbol, 2025-05-22:
Worked like a charm, you the man. This software rocks. 

Have been reading round the site, is there somewhere I can selflearn about the code, what does what and how to contribute?

[image]

dalathegreat, 2025-05-22:
Wow! Very nice 😎 

You can support the project by joining the [Patreon Discord server,](https://www.patreon.com/c/dala) there is where we hangout and discuss, troubleshoot, develop, etc. , it is the best way to get involved with the project!

closed 2025-05-22 by dalathegreat

JeppeDegnbol, 2025-05-23:
Now on both patreon and discord, the system ran flawlessly thru the night, and did a good job of catching me pulling the extra wire I put in with grounded shield back out, to be replaced with the original common cable not grounded but shielded. Whent into error after removing it, after recconection it did not clear its own error state and went into fault and opened the contactors as it shoud, so had to be rebooted. works fine after that.

I will go dig see If I can figure out how the code works, and perhaps see if I can fix those minor issues with not closing contactors again, not catching the can coming back online etc.

PR #1159 (2025-05-22): Bugfix: Zoe gen2 issues
### What
This PR has some general improvements for Zoe Gen2 52kWh

### Why
Fixes #1158 

### How
We apply the following fixes
- We now read all 96 cellvoltages
- Correct ID for CAN message 0x376
- Only read cellvoltage values higher than 500mV (Avoids accidental read of unavailable value)
- Only publish celltemperatures once both values have been updated
- Only publish cellvoltages once both values have been updated