dalathegreat/Battery-Emulator #1171: [BUG]

pa5zl, 2025-05-24:
**Describe the bug**
the CPU temp shown in the webpage and also in the mqtt topic seems to be "stuck" at 53.3deg.
every xx seconds it shows the "actual" temp which is around 42 deg and the it jumps back to 53.3deg

**Expected behavior**
show the actual cpu temp without these "jumps"

**Screenshots**

[image]

[image]

[image]

[image]

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: 8.13.0
- Battery used: [FAKE]
- Inverter communication protocol: [BYD_MODBUS]
- Hardware used for Battery-Emulator: [HW_LILYGO]
- MQTT: [yes]

mosquitto mqtt broker on local lan and using mqtt explorer to test.
also a ping runs to the liliygo from a zabbix server for (device)monitoring 

NO Battery of invertor connected(yet) only testing

powered by a 2amp 5v phone charger or a 12volt psu  , behavoir is the same.

RogMoe, 2025-05-24:
8.13 works fine on my Lilygo/Fronius/Leaf pack, both webserver and MQTT.
Reading is accurate.

dalathegreat, 2025-05-25:
Tested on 3 boards, unable to reproduce this issue 🤔

dalathegreat, 2025-06-17:
Closing this as unable to reproduce, now seen over 10 boards without this issue