dalathegreat/Battery-Emulator #1187: Configurable CPU Temperature Threshold for EVENT_CPU_OVERHEATING

cray9000, 2025-05-29:
**### **Configurable CPU Temperature Threshold for EVENT_CPU_OVERHEATING****

When using my Liligo ESP32 device, powered via the USB port of a Fronius Gen24 inverter (in the connection area), the ESP32 CPU occasionally reaches temperatures around 78–81 °C under load. In this setup, the fixed CPU temperature threshold frequently triggers the EVENT_CPU_OVERHEATING, resulting in excessive event logs via MQTT.

To resolve this, I made the CPU temperature limit for the first overheating alert configurable via a central definition in user_settings.h.

Implemented Changes (Working as Expected):

**_user_settings.h (line 173):_**
_#define CPU_TEMP_LIMIT_OVERHEATING 83_

_**datalayer.h (line 220):**_
Added a configurable limit field:
typedef struct {
 ...
  _/** ESP32 main CPU temperature, limit for overheating */_
  _float CPU_temperature_limit_overheating = CPU_TEMP_LIMIT_OVERHEATING;_

_**safety.cpp (line 25):**_
Adjusted the check to use the configurable limit:
...
_if (datalayer.system.info.CPU_temperature > datalayer.system.info.CPU_temperature_limit_overheating) {_
  ...
}

Notes:
I cannot guarantee that the variable names and comments fully conform to all code conventions or use cases. However, the implementation has been successfully tested in my setup using a Liligo device, Fronius Modbus, and a Pylontech battery. The system is running reliably.

dalathegreat, 2025-06-01:
Nice! But, this kind of defeats the purpose of the check, to warn the user that you are exceeding max safe temperature limits for longterm stable operation. Instead if you are mounting it in a place with limited airflow, mounting a small heatsink on the CPU might be a better idea than a software configuration option, as explained in the WIki: https://github.com/dalathegreat/Battery-Emulator/wiki/Installation-guidelines#keeping-the-battery-emulator-cooled

dalathegreat, 2025-06-02:
I will close this issue, since I believe we should not add unnecessary parameter tweaking. The warning event is an actual warning, you are running too hot for longterm operation.