dalathegreat/Battery-Emulator #1211: Hallo i have a problem with BYD 60 k , the contract is closed shortly and reopen after one second, in the event log i have "battery is completely discharge "

How can I solved the problem 🥺

alaatamim1996, 2025-06-06:


demon1300, 2025-06-07:
Can you provide some more information on the setup? What voltage is the battery currently at? Could you provide a screenshot of the battery emulator main page and also one from the "cellmonitor" page?

alaatamim1996, 2025-06-07:
Yes of course 
If you need more data tell me 

[image]
[image]
[image]

demon1300, 2025-06-07:
As per the Wiki (https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Deye) what does it say on the inverter for battery statistics when you first start it up and then once it faults?

alaatamim1996, 2025-06-07:
The inverter showed that the battery was connected, the battery protocol and the charge percentage were shown, but the voltage was shown as zero. When measuring the voltage on the battery terminals, the result was zero.

dalathegreat, 2025-06-07:
Focus on getting the battery to output voltage first. Skip the inverter entirely for now.

Sounds like you have a crash locked Atto3 battery https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BYD-Atto-3-%E2%80%90-Yuan-Plus#how-do-i-unlock-a-crashed-battery

demon1300, 2025-06-07:
@dalathegreat was about to suggest the same however the puzzling part is the fact that the contactors close for a second then mentioned they open 🤔

closed 2025-06-11 by alaatamim1996