dalathegreat/Battery-Emulator #1213: [Feature request] LANDROVER VELAR intergration.

Jmieng880, 2025-06-08:
Picked up a cheap velar battery (17480WH) 
I have a fox Inverter
Ive managed to get hold of the CAN database files and selection of drive/charging logs.
Ive measured the CRASH signal and normal operation is 10hz at 12v 50%PWM. 
Ive created Harness for Battery LV and HV and should be able to start testing soon.
Could anyone help integrate?
Is there a project area to upload files to share?
Thanks Paul

bingflash2, 2025-08-12:
please upload your can database & all can log files to google drive, then post the links here.
also your wiring diagram / harness information to assist other users with the same battery.

a separate thread / feature request has been added for remote battery development (can bridge over tcpip), meaning you may in the future be able to wire up your lillygo to your battery, and have the developers use your battery remotely to develop the solution!

taking lots of logs is perfect and its likely your battery will get implemented if you upload all the information you have.

Jmieng880, 2025-08-12:
@bingflash2 thanks here you go.

https://drive.google.com/drive/folders/1xN1zJRIA91bIikV4aqEYttHSR9kUu1g2?usp=sharing

bingflash2, 2025-08-17:
can you please post some high res photos of the connectors & stickers, that way we can create a diagram to wire it up.

Jmieng880, 2025-08-17:
Yes will do. I have also used a signal generator to satisfy the crash
signal ;) will upload soon

Jmieng880, 2025-08-24:
Current HW setup with Schematic

[image]