dalathegreat/Battery-Emulator #1224: [BUG] FOX ESS inverter not showing SOC

sprocketman, 2025-06-10:
**Describe the bug**
A clear and concise description of what the encountered bug is

**To Reproduce**
Please try to explain the steps required to reproduce the issue

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version and settings (please complete the following information):**
- Please attach your USER_SETTINGS files for easier debugging
- Software version: X.Y.Z
- Battery used: [NISSAN_LEAF]
- Inverter communication protocol: [BYD_MODBUS]
- Hardware used for Battery-Emulator: [HW_LILYGO/HW_STARK/HW_DEVKIT]
- CONTACTOR_CONTROL: [yes/no]
- MQTT: [yes/no]

sprocketman, 2025-06-10:
I have an issue but not sure if its a bug or me?

The Fox ESS inverter wont disply SOC from a Liligo board and BMW I3 battery. The events page indicates that there is no comunication with the inverter CAN. This has previously worked when BMW PHEV was selected instead of I3?

[image]

To Reproduce
Please try to explain the steps required to reproduce the issue
The SOC is displayed by the Liligo so it must be an issue with the communication with the Fox inverter?

[image]

Both inverter and Battery show Can and Not Can add on?
Version and settings (please complete the following information):
8,13 version
#define BMW_I3_BATTERY
#define FOXESS_CAN
#define HW_LILYGO
#define CAN_ADDON           
#define CRYSTAL_FREQUENCY_MHZ 8 

Thanks

dalathegreat, 2025-06-11:
Make sure terminating resistors are OK
Make sure configuration is right (I assume MCP2515 chip is connected to inverter?). Please attach the USER_SETTINGS files here.

sprocketman, 2025-06-11:
I wrote too soon,
I am getting this error now.

[image]