dalathegreat/Battery-Emulator #1225: Tesla Model 3 LFP Battery pack unbalanced

1ni3shi, 2025-06-11:
Tesla Highland Model 3 LFP RWD 60kwh
Deye Inverter
Lilygo
BE v8.14 currently (was on v8.12 prior)

My pack has been unbalanced for quite sometime now, currently at around 65-66% true SOC one of the cells will have exceed 3650mv causing the emulator to stop charging and reporting 100% SOC to the inverter.
I have tried to initiate balancing at around 3400mv, but nothing much has changed after 60mins.
If I balance at around 3500mv, it will eventually exceed 3650mv and charging will stop and nothing changes after that.

Is this a known issue with Deye/Tesla setup?

Also with the v8.14 manual balance option is no longer showing under change settings.

Much thanks for any input!

dalathegreat, 2025-06-11:
Tesla batteries are not well supported. We have not gotten balancing to work yet. This is indicated in the Wiki, the Telsa only has 1/3 stars due to this

[image]

Cedar12345, 2025-07-11:
Did you manage to get your Tesla pack back to balance? Interested to know if there is any progress on Tesla battery balancing as these are the most readily available batteries in my area.

1ni3shi, 2025-07-12:
No, balancing does not work. SOC drift will occur over time. Best to just check cell voltage and use voltage instead of SOC%. I don’t rest my pack unless something is wrong with it, which prevents the pack for recalculating SOC %.