dalathegreat/Battery-Emulator #1227: Battery capacity mismatch between Fronius and BE

RogMoe, 2025-06-11:
There seems to be a mismatch between Nissan Leaf overall battery capacity and the battery capacity being sent to the Fronius inverter.
According to Nissan, the 30kWh Leaf pack has a capacity of 28.9kWh. After degradation there is a usable 21.9kWh left in my case. When Fronius polls BE at boot for the battery capacity, it tells Fronius it's the full 28.9kWh.

Mentioned this behaviour to Dala. His reply:
_The Fronius register is read once on startup, before the battery has had time to start and give proper values._

Possible solution could be to store the capacity value in ESP's Flash Memory and updated every (daily BMS) reboot, and send that value when Fronius startup polls BE. Not sure if this issue is the case with other batteries and/or inverters.

[image]

[image]

dalathegreat, 2025-06-16:
Before we do any fixing here, let's wait until we see that it is required. We do not know if the BYD batteries send the maxCapacity, and just scale it with SOH%. Another unknown is, does Fronius even use this info for the ECA? More info needed before we decide to do something!

RogMoe, 2025-06-27:
Got a chance to ask questions during the webinar, but they only answer the easy questions...
So, wait and see I guess.